epfl-lasa/iiwa_ros #11: HOW TO MAKE THE ROBOT MOVABLE?

chunxuli90, 2019-07-09:
Hi,

I am a new user of ROS. Just a quick question, how to make the kuka iiwa robot move by using your repo?

costashatz, 2019-07-09:
> I am a new user of ROS.

Please:

1) spend some time reading the tutorials of ROS,
2) Read the ros_control page: http://wiki.ros.org/ros_control (all of our controllers are based on this pipeline)
3) Start first in simulation (we provide Gazebo) and once everything is working and seems safe, transfer to the robot

> how to make the kuka iiwa robot move by using your repo?

To start moving things in simulation I propose the following:

1) Use position control: change [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/launch/iiwa_gazebo.launch#L38) to `<arg name="controller" default="PositionController"/>`
2) Launch gazebo with iiwa: `roslaunch iiwa_gazebo iiwa_gazebo.launch`
3) Now there should be a topic `/iiwa/PositionController/command` where you can publish desired joint positions and the robot should go to them

chunxuli90, 2019-07-10:
So how is the command like? I have put as: 

rostopic pub /iiwa/PositionController/command std_msgs/Float64MultiArray "data: -0.9 -0.9 -0.9"

It gives me the error:

ERROR: Field [data] must be a list or tuple instead of: str

Args are: [layout.dim layout.data_offset data]

Many thanks.

costashatz, 2019-07-10:
> So how is the command like? I have put as:
> 
> rostopic pub /iiwa/PositionController/command std_msgs/Float64MultiArray "data: -0.9 -0.9 -0.9"
> 
> It gives me the error:
> 
> ERROR: Field [data] must be a list or tuple instead of: str
> 
> Args are: [layout.dim layout.data_offset data]
> 
> Many thanks.

You can use [rqt_gui](http://wiki.ros.org/rqt_gui)...

chunxuli90, 2019-07-10:
Hi,

That is to much for me for now.....

I was trying to make the simulated iiwa robot in gazebo move by typing commands like:
rostopic pub /iiwa/PositionController/command std_msgs/Float64MultiArray "data: [-0.9 -0.9 -0.9]", but it fails, shown below:

rostopic pub /iiwa/PositionController/command std_msgs/Float64MultiArray "data: [-0.9 -0.9 -0.9]"
publishing and latching message. Press ctrl-C to terminate
[WARN] [1562750138.474672]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750138.653897]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750138.755902]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750138.857637]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750138.959838]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750139.061921]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750139.163842]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'
[WARN] [1562750139.265604]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'layout: 
  dim: []
  data_offset: 0
data: [-0.9 -0.9 -0.9]'

Could you please advise the correct format of the commands? Thank you.

costashatz, 2019-07-10:
> That is to much for me for now.....

This is a graphical interface. It's way easier! I never send commands through the command line. Alternatively you can use [python to send commands](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29#rospy_tutorials.2BAC8-Tutorials.2BAC8-WritingPublisherSubscriber.Writing_the_Publisher_Node) (change the names/types of the topics).

> Could you please advise the correct format of the commands?

It is an array of values: [a, b, c, .., d]

chunxuli90, 2019-07-10:
Thank you, I have done it.

closed 2019-07-10 by chunxuli90