epfl-lasa/iiwa_ros #12: Link with moveit to control real kuka iiwa robot

chunxuli90, 2019-07-10:
Hi,

Currently I am already able to send text data based command via rqt-gui to control the real robot using FRI. But I want to do it with visual monitoring. Can this repo mix with moveit to control the real robot by using the motion planning function? I only have the FRI without Serving motors packages in the KUKA side.

Thank you.

closed 2019-07-10 by chunxuli90

costashatz, 2019-07-11:
> Can this repo mix with moveit to control the real robot by using the motion planning function? I

Not yet! But #10 will do exactly this.. I need to find time to review this PR and then I will merge..

costashatz, 2019-07-11:
By the way, you can use rviz to visualize the current state of the robot etc..