epfl-lasa/iiwa_ros #13: joint impedance controller and cartesian impedance controller support

jacknlliu, 2019-09-01:
Is it possible to use the joint impedance control mode and cartesian impedance control mode with this package? And change the stiffness and damping parameters on the fly?

costashatz, 2019-09-01:
> joint impedance control mode

Yes.. This is enabled by default and you can choose between position or torque joint impedance control..

> And change the stiffness and damping parameters on the fly?

This is not possible since this is not allowed via the FRI library. You can alter the [Java files](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/java/FRIOverlay.java) to be able to change them when starting the program (this is already there for the stiffness)..

closed 2020-01-24 by costashatz

matthias-mayr, 2020-06-16:
Just double-checking: Do I get it right from your answer and the repo that there is currently no support for Cartesian impedance control?

JimmyDaSilva, 2020-06-16:
@matthias-mayr It is possible to use the FRI library with the Cartesian impedance controller. That is what I am doing.

But to my knowledge, it is not possible to change between the available controller (joint impedance, cartesian impedance, torque control) on the fly without stopping the JAVA application running on the SmartPad as asked by @jacknlliu

matthias-mayr, 2020-06-16:
What do you mean with "the Cartesian impedance controller"?
I see two possibilities to implement Cartesian impedance control
1. With the KUKA libraries like in [grl](https://github.com/ahundt/grl/blob/bd62058cb3d7fdaf85ea2841d32199389152e7fa/src/java/grl/src/grl/driver/GRL_Driver.java)
2. Or through `ros_control` and sending efforts

But I can't see any of the two ways being implemented already.

By `That is what I am doing` - do you mean that you have it running already?

JimmyDaSilva, 2020-06-16:
I meant that I am using KUKA's Cartesian Impedance controller.
I just had to change the Java file of iiwa_ros a little bit here:
https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/java/FRIOverlay.java#L124

Now I can send joint positions to the robot and still play with the nullspace of the robot.

matthias-mayr, 2020-06-16:
Okay, nice. Maybe it would be nice to create a pull request for it, so it is easier to access for other people.

> But to my knowledge, it is not possible to change between the available controller (joint impedance, cartesian impedance, torque control) on the fly without stopping the JAVA application running on the SmartPad as asked by @jacknlliu

Do I get it right that this switching between modes is implemented in [grl through the flatbuffers connection](https://github.com/ahundt/grl/blob/bd62058cb3d7fdaf85ea2841d32199389152e7fa/src/java/grl/src/grl/driver/GRL_Driver.java#L862)?
In general I would be interested to be able to adapt those parameters while executing.

JimmyDaSilva, 2020-06-16:
> Do I get it right that this switching between modes is implemented in grl through the flatbuffers connection?
> In general I would be interested to be able to adapt those parameters while executing.

I didn't try the grl implementation, but I think this is indeed possible because SmartServo and DirectServo allow it, I think.

But for now, I don't know any way to change the controller (or its parameters) used on KUKA's side at runtime through the FRI library, hence in iiwa_ros.