epfl-lasa/iiwa_ros #14: IP configuration and connection issues

prachandabhanu, 2019-09-10:
Hallo 

I have the following questions regarding the IP config:

1. I am not able to ping the robot with [IPv4](https://github.com/epfl-lasa/iiwa_ros/blob/c75f970cae69eb89a4c1e69f63c83824c549dc3d/iiwa_driver/config/iiwa.yaml#L3) and Robot IP as mentioned here.
2. Do we need to change the IP [here](https://github.com/epfl-lasa/iiwa_ros/blob/c75f970cae69eb89a4c1e69f63c83824c549dc3d/iiwa_driver/java/FRIOverlay.java#L36) to Linux /ROS Master
3. I connected a monitor to KONI and checked the IPv4 configuration of two adopters first one 172.31.1.147 (Here I connect my Sunrise to sync the and load the app, etc) and second IPv4 config 192.168.0.2. In my guess, the second adopter is KONI IPv4 config.
4. Also, I am confused about whether to change the KONI IP in setup.cat. Currently, it is set to 192.168.0.7 

Thank you

ymollard, 2019-09-11:
1. The `.100` IP in my README was a mistake, it's actually `.1`. It's fixed.

2. The `_clientName` from the Java app must exactly match the one of the ROS computer, which must be on the same subnet than KONI.

3. Don't do that. KRC owns the KONI Ethernet port so its IP can be configured from Sunrise. If possible, revert your changes. As I recall, the KONI Ethernet adapter shouldn't be visible in Windows Embedded unless KRC has been setup to release the Ethernet port. IFL-CAMP/iiwa_stack requires to release it, but this stack doesn't work this way. You may want to re-activate the KONI Ethernet from KRC.

4. Yes. I suggest that you change the KONI IP to revert to the default IP: 192..170.10.2. This is not to be achieved by the means of an external monitor but from the Sunrise project itself in StationSetup.cat > Configuration. If you create a new empty project the KONI interface will be given its default IP.

prachandabhanu, 2019-09-13:
Thank you. It works now.

closed 2019-09-13 by prachandabhanu

costashatz, 2019-09-13:
Thanks @ymollard  for handling this issue!