epfl-lasa/iiwa_ros #17: connection issues

921218316, 2019-10-19:
I configure the IP as the steps   [https://github.com/epfl-lasa/iiwa_ros/issues/14](url)，and I can ping the robot, but when I run the app FRIOverlay in SmartPad and with 10s I have run the roslaunch iiwa_driver iiwa_bringup.launch. I get a timeout error, I can't see the [FRI] tab.
[image]

costashatz, 2019-10-19:
Are you sure your dependencies are installed? Can you try launching the gazebo file? `roslaunch iiwa_gazebo iiwa_gazebo.launch` and let us know what you get..

921218316, 2019-10-19:
When I launch the gazebo, it shows that..,
[image]

costashatz, 2019-10-19:
Can I have the terminal output also? Thanks!

921218316, 2019-10-19:
here are terminal outputs
[image]
[image]
[image]
[image]

costashatz, 2019-10-19:
Something un-related to your issue is that you haven't installed/linked properly to the RBDyn library.

Back to your problem, everything seems to be installed correctly. Are you sure that the IPs are correctly set? Please go back to the README and re-do all the steps..

921218316, 2019-10-20:
I re-do all the steps as  README again.The IPs I configure:  Windows laptop is connected on the X66 Ethernet port and has IP 172.31.1.42 mask 255.255.0.0,Linux/ROS laptop is connected on the KONI Ethernet port and has IP 192.170.10.1 mask 255.255.255.0, but It doesn't work anyway

921218316, 2019-10-20:
On the SmartPAD there is no [FRI] tab in the I/O Groups,is this where the problem lay? If so,how to fix it

costashatz, 2019-10-22:
@921218316 I think I know what's happening. The computer of KUKA has two ethernet ports. One that you should connect in Windows to upload the apps etc., and one for the FRI communication. Are you using this one when connecting to iiwa_ros?

921218316, 2019-10-22:
I don't connect as you mentioned above. I do it as the following picture shows
[image]

costashatz, 2019-10-22:
You should connect your linux machine to the KONI or KOIN (no idea what is correct) ethernet port.. Also double check all your IPs..

921218316, 2019-10-22:
I'm sorry for the wrong picture.It is actually like this.
[image]

 I checked the IPs,it is correct as the picture shows.I can ping on the ros laptop to the sunrise cabinet.And I can also ping on the workbench laptop to the sunrise cabinet

921218316, 2019-10-30:
I first start the ros controller by the command line `roslaunch iiwa_driver iiwa_bringup.launch` and then start the appliaction in the SmartPAD,but it doesn't work.And there is no [FRI] tab the I/O groups .
[image]
[image]
[image]

costashatz, 2019-10-30:
> And there is no [FRI] tab the I/O groups .

There shouldn't be any.. We do not have and it works..

Please make sure that everything is connected and with proper IPs..

921218316, 2019-11-04:
Thank you! I re-assign the port to RTOS:
`C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS`
And it works now!
Thank you very much again!

costashatz, 2019-11-04:
Great news! Closing the issue! Feel free to re-open a new issue if something arises!

closed 2019-11-04 by costashatz

emielke12, 2021-04-16:
I'm experiencing the same issue here, the Timeout is occuring sporadically. Sometimes I can connect, and others not, without changing anything between launches. Any ideas?

matthias-mayr, 2021-04-16:
I had this issue as well and it got a lot better when using it with #47 and #48, especially when using the latter one.

emielke12, 2021-04-16:
@matthias-mayr I'll try them out. Thanks for pointing me to those!

mf093087, 2021-08-22:
> Thank you! I re-assign the port to RTOS:
> `C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS`
> And it works now!
> Thank you very much again!

May I ask, what do you mean by 're-assign the port to RTOS:
> `C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS` . Is it in ubuntu or in windows?

mf093087, 2021-08-22:
> I had this issue as well and it got a lot better when using it with #47 and #48, especially when using the latter one.

Hi, Matthias, I tried your solution but it still doesn't work, I still receive the error timeout before FRI connection....

emielke12, 2021-08-23:
@mf093087 That's a .exe, so it is Windows. I believe this is done on the cabinet PC.

mf093087, 2021-09-04:
May I ask, after re-assign the port to RTOS:
C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS

Should we turn off the kuka controller and remove the external monitor, and then use the AUT mode on smartpad? Can we use AUT mode on the external monitor with windows?

CodingCatMountain, 2023-04-20:
> @mf093087 That's a .exe, so it is Windows. I believe this is done on the cabinet PC.

Hi, @emielke12 . Have you solved this connection problem?? Actually this comments: 

` C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS`, it should be run on the windows 7 which installed on the computers. However, in our group, we also try this command which have to get into the windows 7 in the controller.. the [FRI] tab still is missing..