epfl-lasa/iiwa_ros #18: control iiwa 14 with torquecontrol mode

921218316, 2019-11-12:
1. 
I try to control the real robot with the torquecontrol mode.When I public the message through the command line：`rostopic pub -1 /iiwa/PositionController/command std_msgs/Float64MultiArray "layout:
  dim:
  - label: ''
    size: 0
    stride: 0
  data_offset: 0
data:
- -0.8 -6.1 0. 5.8 -0.3 -0.3 -0.1`
It doesn't move and the terminal shows `[WARN] [1573559821.210509]: Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'label: ''
size: 0
stride: 0'`
I don't know why.What's more,is there any different with the input data:-0.8 -6.1 0. 5.8 -0.3 -0.3 -0.1 and the Joint Torques data show on the smartPAD?

2. 
Can you provide an example to control the real robot with the torquecontrol mode.I will apprivate it !

costashatz, 2019-11-13:
Thanks for using our code.

The controllers are not intended to be used in the command line! This is dangerous as it is difficult to be reactive in the command line. We strongly encourage you to write your controller command as a separate node.

iiwa_ros uses ROS control to interact with ROS and the robot. To understand how ROS control works please have a look at the following:

- http://wiki.ros.org/ros_control
- https://roscon.ros.org/2014/wp-content/uploads/2014/07/ros_control_an_overview.pdf
- http://gazebosim.org/tutorials/?tut=ros_control
- https://sir.upc.edu/projects/rostutorials/10-gazebo_control_tutorial/

Note that you do not have to create the hardware layers or the drivers (we have already implemented that in iiwa_ros), but this would be give you a much better idea on how to use any robot that is controlled via ROS control (and thus KUKA iiwa robots usign iiwa_ros). Feel free to pose any questions that are related directly to iiwa_ros.

921218316, 2020-01-16:
I start the robot using the torquecontroller .but one thing confusing me.To tontrol the iiwa14,we have to send torque message 
/iiwa/TorqueController/command std_msg/float64MultiArray.
What does the command torque mean？Actual torque or external torque？It seems neither

what’s more，you creat a custom_effort_controller, It seems you don't use it

costashatz, 2020-01-16:
> What does the command torque mean？

The torque command is the torque applied directly to the joints. In practice, KUKA always adds the gravity compensation. So:

- τ_msg: torque that you send via `/iiwa/TorqueController/command std_msg/float64MultiArray`
- τ_gc: torque to compensate gravity
- τ: total torque applied to the robot

τ = τ_gc + τ_msg

So you only get to control τ_msg. I hope this is clear.

> what’s more，you creat a custom_effort_controller, It seems you don't use it

The custom_effort_controller is optional. You can choose to use it for advanced things. Regretfully, the documentation is not there yet, so I would tell you to not use it for now if you do not understand what it is doing.

jacknlliu, 2020-01-21:
@costashatz  What's the relationship between the torque command by our custom ros controller and the [stiffness values](https://github.com/epfl-lasa/iiwa_ros/blob/afca61f9dcef8c58d913140bc10a2d813bf78c77/iiwa_driver/java/FRIOverlay.java#L122)? Does the stiffness values still affect our torque controller?

costashatz, 2020-01-22:
> Does the stiffness values still affect our torque controller?

Yes. I do not know the exact relationship (KUKA does not say exactly what they do), but this paper sheds a bit of light: https://ieeexplore.ieee.org/abstract/document/7989213

In any case, if you want pure torque-control, you should set the `stiffness` and the `damping` values to zero. This will make it a bit harder to control in gravity compensation mode though.

jacknlliu, 2020-01-22:
> In any case, if you want pure torque-control, you should set the stiffness and the damping values to zero. 

It's good for using the pure torque control interface.

closed 2020-01-24 by costashatz

mf093087, 2021-09-06:
> The torque command is the torque applied directly to the joints. In practice, KUKA always adds the gravity compensation. So:
>
>     * τ_msg: torque that you send via `/iiwa/TorqueController/command [...]

HI Costas, if we don't use command line, what else should we use? Should we use rqt?

costashatz, 2021-09-22:
> Should we use rqt?

Yes rqt gui is better than the command line. Also making a simple python or C++ script is even more desirable.

XinyuanZhao, 2023-02-04:
> Note that you do not have to create the hardware layers or the drivers (we have already implemented that in iiwa_ros), but this would be give you a much better idea on how to use any robot that is [...]

Hello @costashatz 
I'd like to inquire of the implementation of the hardware interface you provided in `iiwa_ros`. I skimmed the code of `iiwa.h` and `iiwa.cpp` but I didn't find the implementation of functions `read()` and `write()`, which were declared as virtual functions in the class `RobotHW`. Instead, you defined another pair of functions named by `_read()` and `_write()` for the same functionality.

I didn't create my own robot abstract using ROS Control before so I am a bit confused here. In my opinion, ROS Control expects users to override their functions so as to be used in conjunction with the other standard interfaces (e.g. of controllers). Since different interfaces are used, how can the hardware interface implemented in `iiwa_ros` work together with the other components of ROS Control?