epfl-lasa/iiwa_ros #19: add publishing the external wrench applied on end-effector

jacknlliu, 2019-12-24:


costashatz, 2019-12-24:
@jacknlliu thank you for using our software. Can you make your request a bit more specific? The external wrench requires some modeling or maybe a force-torque sensor on the end-effector.. What is your usage exactly and which feature is missing?

jacknlliu, 2019-12-28:
just like the [iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack) using the [CartesianWrench](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_msgs/msg/CartesianWrench.msg) to publish the wrench.

costashatz, 2019-12-30:
> just like the [iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack) using the [CartesianWrench](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_msgs/msg/CartesianWrench.msg) to publish the wrench. It is useful to detect the force and torque just using the robot not other sensors.

`iiwa_stack` is using the internal model of the KUKA robot to get this. I can try to make some modeling for this. What is your exact usage of this value?

jacknlliu, 2020-01-02:
I hope to use it as a wrench sensor.

jacknlliu, 2020-01-09:
@costashatz  reference to https://github.com/CentroEPiaggio/kuka-lwr/issues/38#issue-90917096

costashatz, 2020-01-09:
> It only needs the `getExternalTorque()` and Jacobian transpose matrix.

Indeed FRI provides this functionality and thus it should be easy to do this (although the end-effector calibration will be much more important now). I will try to code this as soon as possible.

CKowal, 2020-01-24:
I was not able to test on the real robot yet but am I right in the assumption that there is no publisher for external joint torques yet implemented in iiwa_ros?
If one wants to add this functionality, it's necessary to start in "FRIOverlay.java", right?

costashatz, 2020-01-24:
> I was not able to test on the real robot yet but am I right in the assumption that there is no publisher for external joint torques yet implemented in iiwa_ros?

I am working on that. Still didn't find time to finish it.

> If one wants to add this functionality, it's necessary to start in "FRIOverlay.java", right?

No no this is not needed. We are using the FRI interface and thus the C++ FRI code give us access to these quantities. Just didn't find time to expose them yet..

I will try to have this by the end of next week (at least an initial implementation)..

jacknlliu, 2020-01-27:
@CKowal This can be implemented in the  [iiwa_driver](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/src/iiwa.cpp)  by the FRI interface, and no need to modified the `FRIOverlay.java`.

CKowal, 2020-02-03:
Any update on this issue or is it possible to help you in any way, @costashatz ?

costashatz, 2020-02-03:
> Any update on this issue or is it possible to help you in any way, @costashatz ?

I did a rather quick implementation. It is on `external_torque` branch. I haven't tested it yet. Can you please try it out and let me know if it is working? If it's working, then I can make it better.

CKowal, 2020-02-04:
> I did a rather quick implementation. It is on `external_torque` branch. I haven't tested it yet. Can you please try it out and let me know if it is working? If it's working, then I can make it better.

It's working, the /external_torque topic seems to publish meaningful data! Awesome work, @costashatz !
Further question: If one wanted to get external cartesian end-effector torques, what would the necessary steps be? Can you recommend a paper or literature for the computation?

costashatz, 2020-02-04:
> Further question: If one wanted to get external cartesian end-effector torques, what would the necessary steps be? Can you recommend a paper or literature for the computation?

What you actually need to do is to compute the Jacobian (you can use iiwa_tools for this), transpose it and take its (pseudo)-inverse (`J_t_pinv`). Then you have: `τ_ext_eef = J_t_pinv*τ_ext_jnt`.

Have a look [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp#L516), [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp#L483-L484), and [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp#L33-L50) to see how to do it more or less using iiwa_ros. I know there is too much code around this file; try to focus on the lines that I sent you.

If you want to know the exact math and analysis, I recommend these [lecture notes](https://ethz.ch/content/dam/ethz/special-interest/mavt/robotics-n-intelligent-systems/rsl-dam/documents/RobotDynamics2017/7-dynamics3_ex.pdf).

CKowal, 2020-02-25:
I tried to implement your (@costashatz ) suggestions and my first test implies that the computed values are not necessarily the right ones.
My result is a 6x7 matrix, the last entry in the first three rows should contain `_τ_ext_eef_` for x,y and z, is this assumption correct?

JimmyDaSilva, 2020-03-09:
@costashatz I can confirm as well that the topic publishes the same values for external torque as displayed on the SmartPad!

Please just note that it is not using the `/iiwa` namespace so it is publishing directly on `/external_torque` 

Many thanks!

closed 2023-04-28 by costashatz