epfl-lasa/iiwa_ros #25: Publish commanded position/torque and interpolated setpoint position

JimmyDaSilva, 2020-03-09:
Hi,

I am using the position control mode and will be needing the resulting commanded torque. 
I think that both the commanded joint position and torque should be published along with the external torque as discussed in #19. The more information, the better.

By replicating what @costashatz did on the `external_torque` branch, I tested the functions:
- `getCommandedJointPosition`
- `getCommandedTorque`
- `getIpoJointPosition` (**could not get this one working**)

@costashatz I don't know what plan you had to make your `external_torque` branch better so I didn't write a PR. But let me know if I can be of any help.

Many thanks,
Jimmy

costashatz, 2020-03-09:
@JimmyDaSilva thanks a lot for the comment.

> The more information, the better.

This is not always true, but in this case I agree with you. I just didn't need these quantities yet in my use-cases.

> * `getIpoJointPosition` (**could not get this one working**)

I have no idea what this is exactly. Why would you need this? What is it this value?

> @costashatz I don't know what plan you had to make your `external_torque` branch better so I didn't write a PR. But let me know if I can be of any help.

I need to finish this branch. I just didn't find time to do that yet.

> If interested, I made small changes to the `external_torque` branch on my fork: https://github.com/JimmyDaSilva/iiwa_ros/tree/external_torque

I will have a look at your code and let you know.

Thanks a lot again!

JimmyDaSilva, 2020-03-09:
> I have no idea what this is exactly. Why would you need this? What is it this value?

From the FRI manual. 
`getIpoJointPosition()`: Interpolated setpoint position as a basis for path superposition.

This is related to the path superposition (or motion overlay) mentioned in the manual.
I am not sure about what this is supposed to give either.

costashatz, 2020-03-09:
> I am not sure about what this is supposed to give either.

I would not output this one then. I want to have code and data that people use, otherwise it is super hard to maintain the package.

> I will have a look at your code and let you know.

I had a quick look at your code. Seems good enough. Can you make a PR but target my `external_torque` branch? And then I will clean-up/make a bit more generic everything and merge..

JimmyDaSilva, 2020-03-09:
Done with #26 
Thank you @costashatz for taking this into consideration.

closed 2020-03-09 by JimmyDaSilva