epfl-lasa/iiwa_ros #28: ClientCommandMode "WRENCH" and "setWrench" in FRI

CKowal, 2020-03-11:
Hey,

right now the ClientCommandModes are limited to "POSITION" and "TORQUE". Having a look at the FRI documentation a "WRENCH" mode does exist and there is also a method in LBRCommand named "setWrench" to apply a translational force along x,y or z.
Are there any limitations why this couldn't work using iiwa_ros or has it simply not been added yet due to missing applications?

Thanks,
Chris

costashatz, 2020-03-11:
Hello! Thanks for using our library.

> Are there any limitations why this couldn't work using iiwa_ros or has it simply not been added yet due to missing applications?

I haven't checked whether it is easy to add this in iiwa_ros, but also I never really had any use-case in my applications.

Do you need this feature?

CKowal, 2020-03-11:
It would be interesting for some of our application but it probably leads us back to the problem of computing robust external wrenches, otherwise I can't comprehend how the FRI is able to assure that a constant force is applied.
Also interesting, would it be possible to continuously change the applied force to have some kind of dynamic behaviour?

costashatz, 2020-03-11:
> It would be interesting for some of our application

If you just need to control in end-effector space, you can have a look at the [CustomEffortController](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp) (I need to write a documentation at some point!).

> Also interesting, would it be possible to continuously change the applied force to have some kind of dynamic behaviour?

Yes it would be possible if we do this properly.

In any case, I will leave this open to enable this in the future.

CKowal, 2020-03-11:
> If you just need to control in end-effector space, you can have a look at the [CustomEffortController](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp) (I need to write a documentation at some point!).
Will have a look at this for now. 

> In any case, I will leave this open to enable this in the future.
Alright!

JimmyDaSilva, 2020-03-12:
I am also very interested in the "WRENCH" mode.
Being able to switch between the modes after the FRI connection as been established would be great as well.

costashatz, 2020-03-12:
> I am also very interested in the "WRENCH" mode.

As I said to @CKowal, have a look at the CustomEffortController. I will try to implement the "wrench" mode as soon as possible.

> Being able to switch between the modes after the FRI connection as been established would be great as well.

This is not possible with the current FRI library of iiwa. Or at least I do not know how to do it. I'd need help on this one.

JimmyDaSilva, 2020-03-12:
> This is not possible with the current FRI library of iiwa. Or at least I do not know how to do it. I'd need help on this one.

After looking at the documentation and example, I also think that it can't be done for now.

matthias-mayr, 2023-01-02:
Quick update: the Cartesian impedance controller package is published here: https://github.com/matthias-mayr/Cartesian-Impedance-Controller
There's also a short paper describing the background: https://arxiv.org/abs/2212.11215

One of its features is to apply wrenches (forces & torques) with the end-effector. Except for the Jacobian, it's an open-loop approach. So from our side we command torques for the joints and the KUKA side takes care of their implementation.

closed 2023-04-28 by costashatz