epfl-lasa/iiwa_ros #29: How to verify the torque controller is correct

921218316, 2020-04-29:
There are two controller: PositionController and TorqueController. The PositionController is simple to understand, but how to verify the torque controller is correct? Is there any method in gazebo simulation environment?

costashatz, 2020-04-29:
> how to verify the torque controller is correct?

What do you mean verify that is correct? The torque controller will just forward the torques to the robot (either simulated or real robot).

921218316, 2020-04-30:
The command /iiwa/TroqueController/command send to the robot in the gazebo, so I want to know the effect of the robot movement，like joint position, joint velocity, joint acceleration,  after the torque command is applied to verify it is crroct

costashatz, 2020-04-30:
> so I want to know the effect of the robot movement，like joint position, joint velocity, joint acceleration, after the torque command is applied

You can use the topic `/iiwa/joint_states` to track the movement of the arm (it gives joint positions, velocities and torques. But this has nothing to do with the package (if it is "correct" or not; I am not even sure what you mean). Gazebo takes the current state, the commanded torque and uses a physics library to simulate the real-world (by default Gazebo use ODE).

I am closing this issue. Feel free to re-open it if you have any further issues/questions.

closed 2020-04-30 by costashatz