epfl-lasa/iiwa_ros #30: Dynamic equation of the iiwa14 robot

921218316, 2020-04-30:
The general dynamic equation of the robot is `M(\Theta )\ddot{\Theta }+C(\Theta,\dot{\Theta })+G(\dot{\Theta })=\tau `.  In order to simulate the iiwa14 robot in gazebo or conrol the robot, we need to establish the dynamic equation. So, in which source file is the dynamic equation of robot expressed explicitly ? If there isn't any, how the dynamic equation is established and control the robot in your code?

JimmyDaSilva, 2020-04-30:
As said @costashatz  in #29, there is no explicit equation of the dynamics written in this package.

**For real torque control**: The internal torque controller of KUKA is confidential and not available, so we don't know how Kuka uses the torque values. Maybe possible to get a guess by looking at their patents...

**For simulation**: It depends on what you use to simulate the robot. If you use `iiwa_gazebo` from this package, Gazebo is going to use the dynamic parameters set in the URDF and the physics solver you use in Gazebo (by default ODE).

See gazebo [documentation](http://gazebosim.org/tutorials/?tut=ros_control).
Also, the gains for the gazebo controller are set [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/config/iiwa_control.yaml)

costashatz, 2020-04-30:
@JimmyDaSilva thanks for the reply!

To complement what you said, in `iiwa_tools` we are using the [RBDyn](https://github.com/costashatz/RBDyn) library to get some parts of the equation (gravity and coriolis forces, and jacobians mostly; if you need more, you can use RBDyn and add them to iiwa_tools, a PR is always welcomed). See [iiwa_tools](https://github.com/epfl-lasa/iiwa_ros/tree/master/iiwa_tools) for more details.

921218316, 2020-05-01:
@costashatz @JimmyDaSilva 
If so, who do we design ourselves' torque controller to control the robot, like adaptive controller, sliding mode controller and so on？

costashatz, 2020-05-01:
> If so, who do we design ourselves' torque controller to control the robot, like adaptive controller, sliding mode controller and so on？

You should create a different ROS node (or even ROS package), you write your controller there (using any library you want or you write the equations of the robot/control), you read the joint states (from the /iiwa/joint_states topic) and you send your torque commands (through the /iiwa/TorqueController/command topic).

921218316, 2020-05-07:
The effort published by the topic of `/iiwa/Joint_states` is the `τ_msg: torque that you send via /iiwa/TorqueController/command std_msg/float64MultiArray` or the `τ: total torque applied to the robot` as you mentioned in #18 ?

JimmyDaSilva, 2020-05-07:
It's `τ: total torque applied to the robot`

`τ_msg` is the desired command sent via the topic `/iiwa/TorqueController/command`

If you plot both, you will see the KUKA black box magic creating the difference between the desired and actual joint torque

costashatz, 2020-05-10:
@921218316 I am closing the issue. Let us know if you have any other question/remark..

closed 2020-05-10 by costashatz