epfl-lasa/iiwa_ros #31: torque information

921218316, 2020-05-28:
- As you mentioned in #18 `τ = τ_gc + τ_msg` , and you implemented gravity compensation [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/src/gravity_compensation_hw_sim.cpp)

>         if (_iiwa_gravity_client.call(_gravity_srv)) {
>            for (size_t i = 0; i < n_dof_; i++) {
>                C[i] = _gravity_srv.response.compensation_torques[i];
>            }
>        }

 I read your code, in iiwa_tool package, you computed the gravity compensation part by
>         rbd::forwardKinematics(rbdyn_urdf.mb, rbdyn_urdf.mbc);
>         rbd::forwardVelocity(rbdyn_urdf.mb, rbdyn_urdf.mbc);
>         fd.computeC(rbdyn_urdf.mb, rbdyn_urdf.mbc);
>        // Get gravity and Coriolis forces
>         return -fd.C();
but the function fd.C return the non linear effect vector (coriolis, gravity, external force), not just gravity.

- Also, we send the torque message by the topic of `/iiwa/TorqueController/command`, the torque is the desired torque which needs some time to achieve or immediate response?

costashatz, 2020-05-28:
> but the function fd.C return the non linear effect vector (coriolis, gravity, external force), not just gravity.

Yes this is what we call gravity compensation: Gravity and Coriolis forces (we cannot compute the external forces, since we are not providing this information to the model).

> * Also, we send the torque message by the topic of `/iiwa/TorqueController/command`, the torque is the desired torque which needs some time to achieve or immediate response?

The response can never be immediate on the real robot. KUKA's torque tracker is quite good and thus it will almost immediately try to track the torque that you send. Of course, KUKA is doing a lot of things in the background which we do not have access nor a lot of information, so I cannot say for sure.

In simulation, the response is immediate, that is in the next step of the physics.

> * Can you provide a test example, such as tracking a planned trajectory, to test the torque controller work correctly?

I can, but not in the near future. Most probably at the end of August I can find some time.

But what exactly are you trying to identify that is working correctly? We have been using the torque controller with the real robot and the Gazebo simulator for quite some time and we did not find any issue with it.

921218316, 2020-05-29:
> But what exactly are you trying to identify that is working correctly? We have been using the torque controller with the real robot and the Gazebo simulator for quite some time and we did not find any issue with it.

when I tracking a planned trajectory, I specified a few points on the path, then I interpolated them for trajectory planning, and  I used inverse dynamics to calculate the corresponding torque. The gravity compensation is done in your hardware abstraction layer code, and the torque calculated by inverse dynamics is the total torque. The calculated total torque is received through the topic` /iiwa/ torquecontroller/command`. So I set the gravity compensation part C [i] in the hardware abstraction layer to zero, and the simulation results are quite different.
[image]
I do this work mainly referencing [this](https://ww2.mathworks.cn/help/robotics/examples/control-lbr-manipulator-motion-through-joint-torque.html?s_tid=answers_rc2-3_p6_Topic&requestedDomain=cn), and I transplanted this idea to your code iiwa_ros. This is why I want to identify that is working correctly.

costashatz, 2020-05-29:
> The calculated total torque is received through the topic` /iiwa/ torquecontroller/command`

First, you should not push the total torque here. The gravity compensation is added automatically in the robot, and if you use the default values for Gazebo, it is also automatically added. You should only send your "control" torque (if your computations include the gravity and coriolis forces, you can remove them from the computed torque to pass the "clean" value).

> So I set the gravity compensation part C [i] in the hardware abstraction layer to zero, and the simulation results are quite different.

I do not get what you mean that you set zero the gravity compensation and the results are different. 

In the example that you sent me, you should transform this line `tau1 = tauFeedForward(h,:);` to `tau1 = tauFeedForward(h,:) - GravityCompensation(q, qDot);` where q and qDot are the measured current joint positions and velocities.

Is this what you are doing?

Moreover, you should tune the PD gains for the feedback control. We might not be using exactly the same model for the IIWA (check our URDF and theirs, the inertia properties might be different).

921218316, 2020-05-29:
> I do not get what you mean that you set zero the gravity compensation and the results are different.
I mean that I gravity compensation in your `GravityCompensationHWSim`, I do this        

>  if (_iiwa_gravity_client.call(_gravity_srv)) {
>             for (size_t i = 0; i < n_dof_; i++) {
>                 // C[i] = _gravity_srv.response.compensation_torques[i];
> 		C[i] = 0;
>             }
>         }
so the gravity compensation is removed, and I push the calculated total torque to the topic `/iiwa/ torquecontroller/command`

and I checked the URDF, the inertia properties is the same.

costashatz, 2020-05-29:
> so the gravity compensation is removed, and I push the calculated total torque to the topic `/iiwa/ torquecontroller/command`
> 
> and I checked the URDF, the inertia properties is the same.

I am not sure that I can help more.

This is surely not a problem from `iiwa_ros`, but from your controller. Setting torques using the command (especially when you set the gravity compensation to zero) is using only core ROS functionality. Additionally, we have been using this for almost a year now without any issue.

Did you try playing with the PD gains? There might be differences in the Gazebo simulator and I am also enforcing all constraints for the simulation (e.g., maximum velocities etc.); I noticed that in other packages they had less maximum velocities than the reported ones (see [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa14.xacro#L38-L39)).

> I mean that I modified gravity compensation in your `GravityCompensationHWSim`, I do this

The proper way of doing this is to remove this [line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa.gazebo.xacro#L33).

921218316, 2020-05-29:
> Additionally, we have been using this for almost a year now without any issue.

In this case, is it convenient to provide some simple application code using torque controller？It will help me a lot, thank you!

costashatz, 2020-05-29:
> In this case, is it convenient to provide some simple application code using torque controller？As I am a new learner, It will help me a lot, thank you!

I will, but as I told you earlier: 
> I can, but not in the near future. Most probably at the end of August I can find some time.

921218316, 2020-06-04:
I want to make sure that if I computed the torque` τ `using Inversedynamics function, I have to remove `gravity and Coriolis forces τ_gc`, the the torque `τ _conrtol = τ -τ_gc` is send to the topic `/iiwa/ torquecontroller/command`. Am I right?

costashatz, 2020-06-04:
> Am I right?

Yes!

921218316, 2020-06-05:
It works now, the  previous issus was caused by velocity constraint defined in urdf. Thank you very much! But I still hope you to provide a test example, such as tracking a planned trajectory, some time in the future. Thank you again.

921218316, 2020-09-27:
I'm sorry to disturb you, do you have time to provide an example, using torque controller, to track a planned trajectory to test the torque controller work correctly?

closed 2022-02-05 by costashatz

XinyuanZhao, 2023-01-30:
> The proper way of doing this is to remove this [line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa.gazebo.xacro#L33).

> First, you should not push the total torque here. The gravity compensation is added automatically in the robot, and if you use the default values for Gazebo, it is also automatically added. You [...]

Hello @costashatz

Thank you very much for the detailed explanations above and in other related issues.

If I understand correctly, the gravity compensation for a real IIWA robot is always added and cannot be switched off. It is set to be automatically running also in simulations so as to achieve consistent performances as in experiments. Whereas we can switch this function off following the *proper way* you mentioned above in simulations. Is this right?

So ... what is the expected usage of the function `iiwa_tools::IiwaTools::gravity()`? Does it mean to be used when I have a proper ID from somewhere else and then remove the gravity compensation part before sending it out through `/iiwa/ torquecontroller/command`?