epfl-lasa/iiwa_ros #32: Control 7th DOF in Position Control Mode

Tomakko, 2020-06-26:
Hi,

thanks for developing the driver, it works smoothly in our lab for the LBR Med!

I was wondering in there is a way to specify the elbow position in your calculation of the IK. Could you comment on that?

Best,
Martin

costashatz, 2020-06-29:
Hello @Tomakko and thanks for using our library. At the moment it is not possible to use several end-effectors (this is what you are asking I guess) in our IK computations. It is certainly doable to do so though. I will try to give it a look during July.

Tomakko, 2020-06-29:
Not quite, i was asking about a method so specify the configuration of the elbow for one end effector pose.

costashatz, 2020-06-29:
How can you that without defining 2 end-effectors (i.e., frame of references with desired poses: one for the elbow and one for the tip)?

JimmyDaSilva, 2020-06-29:
@Tomakko Doing this is not straightforward.
Like @costashatz said, if you really want the elbow at a precise location, one way to do this could be to split the arm in two and call two IK solvers.

In practice, one rarely needs to really fix the height of the elbow's position. What most people want is it to keep the elbow upwards which is easier to do because it adds fewer constraints.

If you just want the IK solver to give you solutions close to the current position, you can try using `trac_ik` instead of KDL to solve the IK problem:
https://bitbucket.org/traclabs/trac_ik/src/HEAD/trac_ik_lib/

`trac_ik` has several solve types and is easy to integrate to MoveIt. The `distance` solve type should be enough to keep the elbow up if you don't push the robot to its limits
```
% The type can be one of the following: 
% Speed: returns very quickly the first solution found
% Distance: runs for the full timeout_in_secs, then returns the solution that minimizes SSE from the seed
% Manip1: runs for full timeout, returns solution that maximizes sqrt(det(J*J^T))
% Manip2: runs for full timeout, returns solution that minimizes cond(J) = |J|*|J^-1|
```

costashatz, 2020-06-29:
> @Tomakko Doing this is not straightforward.

It is fairly straightforward to implement (not sure if it will always provide the best results though!:P), but needs some time for the implementation and verification. We solve the IK with an optimization loop. We need to only add one constraint, and expose the API to `iiwa_tools`.

> `trac_ik`

`trac_ik` is indeed a very good solution for IK. @Tomakko you can switch to this, instead of our `iiwa_tools` IK solver (we put IK just for completeness, we never intended to be a full-featured IK solver). Of course, if this is not urgent, you can always wait for me to improve `iiwa_tools` but I am not sure when I will do it (I have a quite packed next few months!).

Thanks!

JimmyDaSilva, 2020-06-29:
> It is fairly straightforward to implement (not sure if it will always provide the best results though!:P), but needs some time for the implementation and verification. We solve the IK with an optimization loop. We need to only add one constraint, and expose the API to iiwa_tools.

My bad. I didn't really look at iiwa_tools before. This is definitely something interesting to me.
But it is quite hard to understand what you are doing in those lines:
https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_tools/src/iiwa_tools.cpp#L190-L259

I'll be waiting for the next improvements =)

closed 2023-04-28 by costashatz