epfl-lasa/iiwa_ros #33: err in communication between gazebo and moveit

dbdxnuliba, 2020-06-29:
roslaunch iiwa_gazebo iiwa_gazebo.launch
roslaunch iiwa_moveit demo.launch
when I press plan and execute in moveit,

[ERROR] [1593402180.815820616, 72.486000000]: Action client not connected: PositionTrajectoryController/follow_joint_trajectory

[ERROR] [1593404460.733921626, 2339.063000000]: Unable to identify any set of controllers that can actuate the specified joints: [ iiwa_joint_1 iiwa_joint_2 iiwa_joint_3 iiwa_joint_4 iiwa_joint_5 iiwa_joint_6 iiwa_joint_7 ]

costashatz, 2020-06-29:
> roslaunch iiwa_gazebo iiwa_gazebo.launch

Can you give us the (cmd) output of this command? It seems to me that this did not start correctly..

dbdxnuliba, 2020-06-29:
roslaunch iiwa_moveit demo.launch 

Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "xmate3_RCI_proj" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.

SUMMARY
========

PARAMETERS
 * /iiwa/move_group/allow_trajectory_execution: True
 * /iiwa/move_group/capabilities: move_group/MoveGr...
 * /iiwa/move_group/controller_list: [{'default': True...
 * /iiwa/move_group/jiggle_fraction: 0.05
 * /iiwa/move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /iiwa/move_group/max_range: 5.0
 * /iiwa/move_group/max_safe_path_cost: 1
 * /iiwa/move_group/moveit_controller_manager: moveit_simple_con...
 * /iiwa/move_group/moveit_manage_controllers: True
 * /iiwa/move_group/octomap_resolution: 0.025
 * /iiwa/move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /iiwa/move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /iiwa/move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /iiwa/move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /iiwa/move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /iiwa/move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /iiwa/move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /iiwa/move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /iiwa/move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /iiwa/move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /iiwa/move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /iiwa/move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /iiwa/move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /iiwa/move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /iiwa/move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /iiwa/move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /iiwa/move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /iiwa/move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /iiwa/move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/goal_bias: 0.05
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/init_temperature: 10e-6
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/k_constant: 0.0
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/max_states_failed: 10
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/min_temperature: 10e-10
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/range: 0.0
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/temp_change_factor: 2.0
 * /iiwa/move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /iiwa/move_group/planning_plugin: ompl_interface/OM...
 * /iiwa/move_group/planning_scene_monitor/publish_geometry_updates: False
 * /iiwa/move_group/planning_scene_monitor/publish_planning_scene: False
 * /iiwa/move_group/planning_scene_monitor/publish_state_updates: False
 * /iiwa/move_group/planning_scene_monitor/publish_transforms_updates: False
 * /iiwa/move_group/request_adapters: default_planner_r...
[... 84 lines cut ...]
[ INFO] [1593418735.366015430, 38.496000000]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1593418735.366472356, 38.497000000]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1593418735.366528923, 38.497000000]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1593418735.366550362, 38.497000000]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1593418735.366568422, 38.497000000]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1593418735.366583690, 38.497000000]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1593418735.366600956, 38.497000000]: Using planning request adapter 'Fix Start State Path Constraints'
log : loadLibraryForClass , lookup_name is:moveit_simple_controller_manager/MoveItSimpleControllerManager
[... cut ...]
[ INFO] [1593418744.141401266, 47.232000000]: Failed to call service get_planning_scene, have you launched move_group? at /home/dell/git_code/src/robotic_arm/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:491
[ INFO] [1593418744.461589096, 47.551000000]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1593418746.442960729, 49.520000000]: Waiting for PositionTrajectoryController/follow_joint_trajectory to come up
[ERROR] [1593418752.469235509, 55.520000000]: Action client not connected: PositionTrajectoryController/follow_joint_trajectory

dbdxnuliba, 2020-06-29:
[image]

dbdxnuliba, 2020-06-29:
roslaunch iiwa_gazebo iiwa_gazebo.launch 
... logging to /home/dell/.ros/log/1371c8a8-b9e1-11ea-a97e-d0c63738e37f/roslaunch-dell-NUC8i7BEH-15379.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "xmate3_RCI_proj" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://dell-NUC8i7BEH:38235/

SUMMARY
========

PARAMETERS
 * /iiwa/CustomControllers/controllers/LinearDS/params: [1.0, 1.0, 1.0]
 * /iiwa/CustomControllers/controllers/LinearDS/type: LinearDSController
 * /iiwa/CustomControllers/controllers/PD/input: Orientation
 * /iiwa/CustomControllers/controllers/PD/output: Torque
 * /iiwa/CustomControllers/controllers/PD/params: [20.0, 2.0]
 * /iiwa/CustomControllers/controllers/PD/type: PidController
 * /iiwa/CustomControllers/controllers/PassiveDS/params: [200, 200]
 * /iiwa/CustomControllers/controllers/PassiveDS/type: PassiveDSController
 * /iiwa/CustomControllers/joints: ['iiwa_joint_1', ...
 * /iiwa/CustomControllers/params/end_effector: iiwa_link_ee
 * /iiwa/CustomControllers/params/null_space/Kd: 0.1
 * /iiwa/CustomControllers/params/null_space/Kp: 20.0
 * /iiwa/CustomControllers/params/null_space/joints: [0.04475269104532...
 * /iiwa/CustomControllers/params/null_space/max_torque: 10.0
 * /iiwa/CustomControllers/params/space: task
 * /iiwa/CustomControllers/structure/CascadeCtrl: ['LinearDS', 'Pas...
 * /iiwa/CustomControllers/type: iiwa_control/Cust...
 * /iiwa/PositionController/joints: ['iiwa_joint_1', ...
 * /iiwa/PositionController/type: position_controll...
 * /iiwa/PositionTrajectoryController/action_monitor_rate: 30
 * /iiwa/PositionTrajectoryController/constraints/goal_time: 0.5
 * /iiwa/PositionTrajectoryController/joints: ['iiwa_joint_1', ...
 * /iiwa/PositionTrajectoryController/state_publish_rate: 25
 * /iiwa/PositionTrajectoryController/stop_trajectory_duration: 0
 * /iiwa/PositionTrajectoryController/type: position_controll...
 * /iiwa/TorqueController/joints: ['iiwa_joint_1', ...
 * /iiwa/TorqueController/type: effort_controller...
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1/d: 100
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1/i: 0.01
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1/p: 700
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_2/d: 100
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_2/i: 100
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_2/p: 10000
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_3/d: 1
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_3/i: 35
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_3/p: 4500
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_4/d: 2
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_4/i: 60
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_4/p: 5500
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_5/d: 0.01
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_5/i: 30
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_5/p: 1000
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_6/d: 0.01
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_6/i: 0.1
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_6/p: 500
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_7/d: 0.01
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_7/i: 0.1
 * /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_7/p: 100
 * /iiwa/joint_1_position_controller/joint: iiwa_joint_1
 * /iiwa/joint_1_position_controller/pid/d: 80.0
 * /iiwa/joint_1_position_controller/pid/i: 100
 * /iiwa/joint_1_position_controller/pid/i_clamp_max: 10000
 * /iiwa/joint_1_position_controller/pid/i_clamp_min: -10000
 * /iiwa/joint_1_position_controller/pid/p: 800.0
 * /iiwa/joint_1_position_controller/type: effort_controller...
 * /iiwa/joint_2_position_controller/joint: iiwa_joint_2
 * /iiwa/joint_2_position_controller/pid/d: 100.0
[... 100 lines cut ...]
[ INFO] [1593418696.062572978]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1593418696.494887973, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1593418696.516673021, 0.043000000]: Physics dynamic reconfigure ready.
[INFO] [1593418696.562453, 0.086000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1593418696.694198, 0.161000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1593418696.795965173, 0.161000000]: Loading gazebo_ros_control plugin
[ INFO] [1593418696.796214900, 0.161000000]: Starting gazebo_ros_control plugin in namespace: /iiwa
[ INFO] [1593418696.797270287, 0.161000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1593418696.916739168, 0.161000000]: Loaded gazebo_ros_control.
[urdf_spawner-5] process has finished cleanly
log file: /home/dell/.ros/log/1371c8a8-b9e1-11ea-a97e-d0c63738e37f/urdf_spawner-5*.log
[ INFO] [1593418696.939131019, 0.183000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1593418696.962187662, 0.207000000]: Physics dynamic reconfigure ready.
[INFO] [1593418697.214166, 0.458000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1593418697.215255, 0.459000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[... cut ...]
[INFO] [1593418697.458431, 0.700000]: Controller Spawner: Loaded controllers: /iiwa/joint_state_controller, /iiwa/joint_1_position_controller, /iiwa/joint_2_position_controller, /iiwa/joint_3_position_controller, /iiwa/joint_4_position_controller, /iiwa/joint_5_position_controller, /iiwa/joint_6_position_controller, /iiwa/joint_7_position_controller
[INFO] [1593418697.460358, 0.702000]: Started controllers: /iiwa/joint_state_controller, /iiwa/joint_1_position_controller, /iiwa/joint_2_position_controller, /iiwa/joint_3_position_controller, /iiwa/joint_4_position_controller, /iiwa/joint_5_position_controller, /iiwa/joint_6_position_controller, /iiwa/joint_7_position_controller

costashatz, 2020-06-30:
I think you need to change this [line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/launch/iiwa_gazebo.launch#L41) to `<arg name="controller" default="PositionController"/>` because I think that MoveIt requires a controller that accepts position commands.

matthias-mayr, 2020-07-07:
It worked for me when I changed the controller in this line to `PositionTrajectoryController`.

What do you think about a pull request to introduce something like 
```
<include unless="$(arg driver)" file="$(find iiwa_gazebo)/launch/iiwa_gazebo.launch">
     <arg name="controller" value="PositionTrajectoryController"/>
</include>
```
to [demo.launch](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_moveit/launch/demo.launch)?
There is one other thing that could be resolved then. There seem to be some inconsistencies with the `model` parameter. E.g. `demo.launch` excepts `model` to be `iiwa14` whereas `gazebo.launch` expects `14` only.

dbdxnuliba, 2020-07-08:
aflter I change gazebo.launch file :
[image]

roslaunch iiwa_gazebo iiwa_gazebo.launch

while processing /home/dell/ws_iiwa_ros/src/iiwa_ros/iiwa_driver/launch/iiwa_setup.launch:
while processing /home/dell/ws_iiwa_ros/src/iiwa_ros/iiwa_description/launch/iiwaiiwa14_upload.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/dell/ws_iiwa_ros/src/iiwa_ros/iiwa_description/launch/iiwaiiwa14_upload.launch'
The traceback for the exception was written to the log file

costashatz, 2020-07-08:
> There is one other thing that could be resolved then. There seem to be some inconsistencies with the `model` parameter. E.g. `demo.launch` excepts `model` to be `iiwa14` whereas `gazebo.launch` expects `14` only.

Let me check this and I'll get back to you. Two different people wrote the core library and the moveit integration, so there could be a mismatch.

dbdxnuliba, 2020-07-08:
after add 

<include unless="$(arg driver)" file="$(find iiwa_gazebo)/launch/iiwa_gazebo.launch">
     <arg name="controller" value="PositionTrajectoryController"/>
</include>

[INFO] [1594188426.713041, 0.588000]: Loading controller: /iiwa/joint_1_position_controller
[ERROR] [1594188426.760747930, 0.637000000]: Exception thrown while initializing controller /iiwa/joint_1_position_controller.
Could not find resource 'iiwa_joint_1' in 'hardware_interface::EffortJointInterface'.
[ERROR] [1594188426.760811159, 0.637000000]: Initializing controller '/iiwa/joint_1_position_controller' failed
[ERROR] [1594188427.768668, 1.638000]: Failed to load /iiwa/joint_1_position_controller
[INFO] [1594188427.769645, 1.639000]: Loading controller: /iiwa/joint_2_position_controller
[ERROR] [1594188427.812379885, 1.681000000]: Exception thrown while initializing controller /iiwa/joint_2_position_controller.
Could not find resource 'iiwa_joint_2' in 'hardware_interface::EffortJointInterface'.
[ERROR] [1594188427.812443292, 1.681000000]: Initializing controller '/iiwa/joint_2_position_controller' failed
[ERROR] [1594188428.817892, 2.673000]: Failed to load /iiwa/joint_2_position_controller
[INFO] [1594188428.818715, 2.674000]: Loading controller: /iiwa/joint_3_position_controller
[ERROR] [1594188428.880740116, 2.737000000]: Exception thrown while initializing controller /iiwa/joint_3_position_controller.
Could not find resource 'iiwa_joint_3' in 'hardware_interface::EffortJointInterface'.

matthias-mayr, 2020-07-08:
> roslaunch iiwa_gazebo iiwa_gazebo.launch
>
> while processing /home/dell/ws_iiwa_ros/src/iiwa_ros/iiwa_driver/launch/iiwa_setup.launch:
> while processing [...]

In order to make the setup work for you, the only thing you need to change is [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/launch/iiwa_gazebo.launch#L41) to contain
```
<arg name="controller" default="PositionTrajectoryController"/>
```
as an alternative you can also run
```
roslaunch iiwa_gazebo iiwa_gazebo.launch controller:=PositionTrajectoryController
```

There are not changes needed regarding the `model` parameter.

costashatz, 2022-02-05:
@dbdxnuliba did the fix by @matthias-mayr worked for you?

closed 2023-04-28 by costashatz