epfl-lasa/iiwa_ros #34: how to use iiwa_tool service

921218316, 2020-07-03:
I try to use your code regarding `FK`, but something goes wrong. 

1. 

I read your code `iiwa_service.cpp`, however I don't know waht you mean [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_tools/src/iiwa_service.cpp#L79). What is the meaning of expression `request.joints.layout.dim[0].size`. And which correct value should it be set to

2. 

I start up the `iiwa_gazebo.launch`, then I subscribed the topic `/iiwa/joint_states`, then I I wrote the following code

> class IiwaEeClient
> {
> public:
>     IiwaEeClient()
>     {
>         sub_ = n_.subscribe("/iiwa/joint_states",100, &IiwaEeClient::subStateCallback, this);
>         client = n_.serviceClient<iiwa_tools::GetFK>("/iiwa/iiwa_fk_server");
>     }
> 
>     void subStateCallback(const sensor_msgs::JointState::ConstPtr& msg)
>     {
>         jointstate.name = msg->name;
>         jointstate.header = msg->header;
>         jointstate.position = msg->position;
>         jointstate.velocity = msg->velocity;
>         jointstate.effort = msg->effort;
> 
>         // ROS_INFO("The 6th joint %f", jointstate.position[5]);
> 
>     }
> 
>     void getIiwaEeState()
>     {
>         iiwa_tools::GetFK srv;
>         srv.request.joints.layout.dim.resize(2);
>         srv.request.joints.layout.data_offset = 0;
>         // srv.request.joints.layout.dim[0].label = "height";
>         srv.request.joints.layout.dim[0].size = 1;
>         srv.request.joints.layout.dim[0].stride = _n_joints;
> 
>         // srv.request.joints.layout.dim[1].label = "width";
>         srv.request.joints.layout.dim[1].size = _n_joints;
>         srv.request.joints.layout.dim[1].stride = 0;
>         srv.request.joints.data.resize(_n_joints);
> 
>         for(int point = 0; point < 1; ++point)
>         {
>             for(size_t i = 0; i < _n_joints; i++)
>             {
>                 srv.request.joints.data[i] = jointstate.position[i];
>                 ROS_INFO("The %dth position is %f ",i ,srv.request.joints.data[i]);
>                 // srv.request.joints.data.push_back(jointstate.position[i]);
>             }
>         }
> 
>         if(client.call(srv))
>         {
>             ROS_INFO("The iiwa_ee_state X: %f Y: %f, Z: %f", srv.response.poses[6].position.x, srv.response.poses[6].position.y, 
srv.response.poses[6].position.z);
>         }
>         else
>         {
>             ROS_ERROR("Failed to call service /iiwa/iiwa_fk_server");
>         }
>     }
> private:
>     ros::NodeHandle n_;
>     ros::Subscriber sub_;
>     ros::ServiceClient client;
>     sensor_msgs::JointState jointstate;
>     size_t _n_joints = 7;       // the number of the joint
> 
> };
> int main(int argc, char **argv)
> {
>   //Initiate ROS
>   ros::init(argc, argv, "get_iiwa_ee_state");
>  
>   //Create an object of class SubscribeAndPublish that will take care of everything
>   IiwaEeClient iiwa_ee_client;
>   iiwa_ee_client.getIiwaEeState();
>   // iiwa_ee_client.print_joint_states();
>  
>   ros::spin();
>  
>   return 0;  
> }

but It goes wrong, how should i modify

3. 

I think forwardkinematic is using a set of joint angle as input, and compute the poes of the endeffort, I also try `rqt_service_caller` tools  as following and call it
[image]
but it doesn't work correctly.

costashatz, 2020-07-10:
> but It goes wrong, how should i modify

What is the error that you get? The way you are mixing things up, it seems to me that you are getting some segfault because of unitialized variables or because you are altering the `joint_states` while you are reading them.

You might need to insert a [waitForExistence()](http://docs.ros.org/hydro/api/roscpp/html/classros_1_1ServiceClient.html#a530e3f1d55cf50ab00b8e5bbd9e8230a) call, because sometimes the service takes a while to be visible.

closed 2022-02-05 by costashatz

anubhav-dogra, 2023-01-12:
was there any solution derived for this issue @NKU-ZiXuan  ?  
I am also trying to do the similar things. 
I am facing issues on how to use iiwa_tools. Do I need to subscribe to /iiwa/joint_states and use the `position` messages for the `service.request.joints.data`, like this? but then Im getting segmentation core dumped error.

costashatz, 2023-01-12:
@anubhav-dogra @NKU-ZiXuan this might be related to this https://github.com/epfl-lasa/iiwa_ros/pull/97. Can you try pulling and re-installing everything from scratch (all deps of iiwa_ros)? Make sure to apply the git patch to kuka_fri (see [here](https://github.com/epfl-lasa/iiwa_ros#kuka-fri-private)).

anubhav-dogra, 2023-01-12:
@costashatz Thank you for quick reply. 
All my packages are up-to-date still the problem persists. I think something is wrong in my code for calling the services. 
Can you please suggest a correct way to call these services in a ROS Node. ?

costashatz, 2023-01-12:
#97 (and the related https://github.com/epfl-lasa/robot_controllers/pull/6) was merged 9 hours ago! :) Just making sure you did include those in your latest pull ;) I will try to provide an example as soon as possible.

costashatz, 2023-01-12:
Here's a quick example in python that showcases the usage of `iiwa_service` (not tested on the robot :)). The code in C++ is similar. The code assumes you have launched the robot or the gazebo sim with the `PositionController` (not the default `TorqueController`). This has nothing to do with the `iiwa_service`, but with the code that I am sending commands to the robot.

```python
import rospy
from std_msgs.msg import Float64MultiArray, MultiArrayLayout, MultiArrayDimension
from sensor_msgs.msg import JointState
from geometry_msgs.msg import Pose
import tf
from iiwa_tools.srv import GetIK, GetFK
import numpy as np
from angles import shortest_angular_distance, shortest_angular_distance_with_limits

# Array/list to read joint values from robot
joints = [0, 0, 0, 0, 0, 0, 0]

# Callback for reading joint values
def joint_callback(data):
    global joints
    joints = data.position

# Initialize ROS node
rospy.init_node('test')

# Let's make a publisher to control the robot
pub = rospy.Publisher('/iiwa/PositionController/command', Float64MultiArray, queue_size=10)
# Subscribe to /iiwa/joint_states with a callback (joint_callback)
rospy.Subscriber("/iiwa/joint_states", JointState, joint_callback)

# Let's create a message for command sending
cmd_msg = Float64MultiArray()

# tf listener
listener = tf.TransformListener()

# Names of IK/FK services
ik_service = '/iiwa/iiwa_ik_server'
fk_service = '/iiwa/iiwa_fk_server'

# Wait for services to spawn (it might take a while)
rospy.wait_for_service(ik_service)
rospy.wait_for_service(fk_service)

# Read current pose to set it as target
# CAUTION: This is a dummy valid target. Set your translation/rotation to your desired target!
trans_desired = [0, 0, 0]
rot_desired = [0, 0, 0, 0]

good = False
while not good:
    try:
        # from iiwa_link_0 to iiwa_link_ee
        (trans_desired, rot_desired) = listener.lookupTransform('/iiwa_link_ee', '/iiwa_link_0', rospy.Time(0))
        good = True
    except (tf.LookupException, tf.ConnectivityException, tf.ExtrapolationException):
        rospy.sleep(1.)

# Query IK to go to that location
pose_msg = Pose()
pose_msg.position.x = trans_desired[0]
pose_msg.position.y = trans_desired[1]
pose_msg.position.z = trans_desired[2]
pose_msg.orientation.x = rot_desired[0]
pose_msg.orientation.y = rot_desired[1]
pose_msg.orientation.z = rot_desired[2]
pose_msg.orientation.w = rot_desired[3]

# Print for debugging
# print(pose_msg)

solution = []
good = False
while not good:
    # call the IK service
    try:
        get_ik = rospy.ServiceProxy(ik_service, GetIK)
        seed = Float64MultiArray()
        seed.layout = MultiArrayLayout()
        seed.layout.dim = [MultiArrayDimension(), MultiArrayDimension()]
        seed.layout.dim[0].size = 1
        seed.layout.dim[1].size = 7
        # This is not necessarily needed, it gives an initial guess of the possible joint solution
        seed.data = [0., 1., 0., -1., 0., 1., 0.]

        # Let's get the result from IK
        resp1 = get_ik(poses=[pose_msg], seed_angles=seed)
        solution = resp1.joints.data
        good = True
    except rospy.ServiceException, e:
        print "Service call failed: %s"%e

# # Let's call FK with the solution to check if it is a good one; just for verification
# good = False
# while not good:
#     # call the FK service
#     try:
#         get_fk = rospy.ServiceProxy(fk_service, GetFK)
#         seed = Float64MultiArray()
#         seed.layout = MultiArrayLayout()
#         seed.layout.dim = [MultiArrayDimension(), MultiArrayDimension()]
#         seed.layout.dim[0].size = 1
#         seed.layout.dim[1].size = 7
#         seed.data = solution
#         resp1 = get_fk(joints=seed)
#         sol_pose = resp1.poses[0]
#         print('sol:', sol_pose)
#         good = True
#     except rospy.ServiceException, e:
#         print "Service call failed: %s"%e

print('IK solution:', solution)
target = solution

# Robot limits
lower_limits = -np.array([170, 120, 170, 120, 170, 120 ,175])
lower_limits = lower_limits*np.pi/180.
upper_limits = np.array([170, 120, 170, 120, 170, 120 ,175])
upper_limits = upper_limits*np.pi/180.

# Control the robot to go the computed position (This assumes that the robot is in position control)
while not rospy.is_shutdown():
    # copy global joints into temporal variable
    jt = joints

    # Compute differences of current joint positions to the acquired from IK solution
    diff = []
    for i in range(len(target)):
        diff.append(shortest_angular_distance_with_limits(jt[i], target[i], lower_limits[i], upper_limits[i])[1])

    # Small trick to avoid noisy or very fast movements
    min_threshold = 1e-3
    max_dist = 0.03
    for i in range(len(diff)):
        # Do not allow very big joint commands
        if diff[i] > max_dist:
            diff[i] = max_dist
        elif diff[i] < -max_dist:
            diff[i] = -max_dist
        # Do not allow too small joint commands
        if np.abs(diff[i]) < min_threshold:
            diff[i] = 0

    # Joint position target: current joint + modified differences
    joint_target = np.array(jt) + np.array(diff)

    # Send the data to the robot to follow
    cmd_msg.data = [joint_target[0], joint_target[1], joint_target[2], joint_target[3], joint_target[4], joint_target[5], joint_target[6]]
    pub.publish(cmd_msg)
    rospy.sleep(0.005)

rospy.spin()
```

anubhav-dogra, 2023-01-12:
Awsome @costashatz Thank you so much :)

I will check this code in a while. 

I pulled the latest updates just now. But Im getting the same error which was there earlier with SIMD ON. 
`[ INFO] [1673555981.554147954]: Received urdf from param server, parsing...
[iiwa/iiwa_service-2] process has died [pid 25980, exit code -11, cmd /home/anubhav/stuff_ws/devel/lib/iiwa_tools/iiwa_service __name:=iiwa_service __log:=/home/anubhav/.ros/log/3cce614e-92b9-11ed-ad6f-cbe784b741d0/iiwa-iiwa_service-2.log].
log file: /home/anubhav/.ros/log/3cce614e-92b9-11ed-ad6f-cbe784b741d0/iiwa-iiwa_service-2*.log
`
I even applied the patch with kuka_fri. 
have you also updated your kuka_fri package recently. I have the one from last year. 
if so, let me email you again to get the latest one.

matthias-mayr, 2023-01-12:
It could be that the sizes of the multiarray are wrong. These are code snippets that I used to send commands:
```cpp
std_msgs::Float64MultiArray commands;
commands.layout.dim.resize(2);
commands.layout.data_offset = 0;
commands.layout.dim[0].size = 1;
commands.layout.dim[0].stride = 7;
commands.layout.dim[1].size = 7;
commands.layout.dim[1].stride = 0;
commands.data.resize(1 * 7);
```

```cpp
void overwrite_commands_with_pos(std_msgs::Float64MultiArray& commands) {
    for (size_t joint = 0; joint < 7; ++joint) {
            set_multi_array(commands, 0, joint, _real_states[joint]);
    }
}
```

```cpp
// Helper function to set value in a MultiArray
void set_multi_array(std_msgs::Float64MultiArray& array, size_t i, size_t j, double val)
{
    assert(array.layout.dim.size() == 2);
    size_t offset = array.layout.data_offset;

    array.data[offset + i * array.layout.dim[0].stride + j] = val;
}
```