epfl-lasa/iiwa_ros #36: How to set up properly for realtime capabilities ?

JimmyDaSilva, 2020-07-15:
Hi everyone,

I've been experiencing some "noises" or "jumps" on the robot's side when sending slow and smooth commands, in Position control mode.
I assume these jumps come from a bad communication quality because the noise and jumps get amplified (or more frequent) when I intentionally send commands at 199Hz instead of 200Hz.

I started to look at what UR and Franka do in their ROS drivers.
It seems they both recommend to use a realtime linux kernel, with PREEMPT-RT, and give a simple tutorial:
[Universal Robot : Setting up Ubuntu with a PREEMPT_RT kernel](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md
)
[Franka Emika : Setting up the real-time kernel](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel)

So I have installed a RT kernel as well, and things are a little bit smoother now.
But I am still experiencing some jumps from time to time. And I am not sure if my ros nodes are using the full realtime capabilities of my kernel.

By looking a little bit further into their code, I see that it is required as well to change the thread priority higher by setting the scheduling to SCHED_FIFO:
[UR : `pthread_setschedparam(this_thread, SCHED_FIFO, &params)`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/870d314d65bc3bf85d68764f1f07ccefccc7b140/ur_robot_driver/include/ur_robot_driver/comm/pipeline.h#L349-L394)
[FRANKA : `const int thread_priority = sched_get_priority_max(SCHED_FIFO)`](https://github.com/frankaemika/libfranka/blob/06ad8dcf5706f00663b6fd6351734096cea9c2d0/src/control_tools.cpp#L64-L81)

Now I am wondering how to do this properly in the context of this package...
`iiwa_ros` uses ROS `controller_manager` which seems to be hard-realtime capable as said here: http://wiki.ros.org/controller_manager
I guess we should only have to set the scheduling to SCHED_FIFO in `iiwa_driver`'s thread?
But again I am quite new to these realtime aspects and thread priorities...

In case you are experiencing issues like I do, or have already solved them, I would greatly appreciate if you could reach out to me.
@costashatz I don't know how you are using `iiwa_ros` and if you require realtime control as I do, but would you be interested in improving this package in that direction?

Truly, I am really curious about what you iiwa_ros users have to say and how you control your IIWA.

Cheers,
Jimmy

JimmyDaSilva, 2020-07-16:
Also, I see in this non-ROS driver [`drake-iiwa-driver`](https://github.com/RobotLocomotion/drake-iiwa-driver) that they seem to do something like it [here](https://github.com/RobotLocomotion/drake-iiwa-driver/blob/master/kuka-driver/kuka_driver.cc#L608-L624)

costashatz, 2020-07-18:
> I've been experiencing some "noises" or "jumps" on the robot's side when sending slow and smooth commands, in Position control mode.

I, personally, have never experienced this kind of behavior. However, I was mostly using the robot in torque mode. Do you have a video of this?

> By looking a little bit further into their code, I see that it is required as well to change the thread priority higher by setting the scheduling to SCHED_FIFO:
> [UR : [...]

Yes indeed, we need something like this.

> I would greatly appreciate if you could reach out to me.
> @costashatz I don't know how you are using `iiwa_ros` and if you require realtime control as I do, but would you be interested in improving this package in that direction?

I have never really needed realtime kernel. I was mostly operating the robot at 200Hz-500Hz in torque control with a dedicated PC. I was almost always using another PC for all the other computations. In any case, I think we should improve the package in this direction.

> Now I am wondering how to do this properly in the context of this package...
> `iiwa_ros` uses ROS `controller_manager` which seems to be hard-realtime capable as said here: [...]

As I said apart from your issues, it is good for the package to include real realtime capabilities for those that might actually need it. Let me try to find some time to give it a look. @nash169 can you also share your experience with this?

JimmyDaSilva, 2020-07-22:
> I, personally, have never experienced this kind of behavior. However, I was mostly using the robot in torque mode. Do you have a video of this?

Here is a plot that should explain the issue:
[image]
In this plot, I am moving the 4th axis in position control mode with a 200Hz control (5ms period in FRI), with a slow sinus (20s period).
It works quite well at the beginning. Then you can see some spikes in the torques and the noise gets amplified.
In the end, it gets quite jerky and almost out of control.

The slower you go, the worst it gets.
The more you increase the control frequency, the worst it gets.
It is also better now that I am using Preempt-RT in my Linux kernel. I also dedicated a processor core entirely to iiwa_driver to make sure it does not get disturbed by other processes.

I am planning to make some tests without the ROS layer, and play with the KUKA example LBRJointSineOverlay, to make sure this does not happen also outside of this package.

JimmyDaSilva, 2020-07-22:
[image]
It's quite random. Here I got only one "jump"

costashatz, 2020-07-22:
> position control mode

You are using position control provided from KUKA? Or your own PID-gains? If this comes from KUKA, what gains did you set for "stiffness" and "damping" (see java files)?

Btw, from the plots it seems that the position is being tracked very well, no? There's almost no noise in the position. Are these jumps noticeable while operating the robot? Or only via these plots?

To be honest, we never really experienced this behavior. We would only get some small flickering (in position) in the 7th joint (which is normal given the 200Hz control and the fact that the calibration for the end-effector is not perfect) and we solved this by clamping the maximum command given to the joint (we were using our own controller, not the position control provided by KUKA).

JimmyDaSilva, 2020-07-22:
> You are using position control provided from KUKA? Or your own PID-gains? If this comes from KUKA, what gains did you set for "stiffness" and "damping" (see java files)?

I have modified your Java file to not use JointImpedanceControl but KUKA's JointPositionControl instead.
You can only specify the gains for JointImpedanceControl or CartesianImpedanceControl.
The JointPositionControl I am using is set to be really stiff (much more than the max allowed by JointImpedanceControlMode), but we don't know the gains KUKA is using.

> Btw, from the plots it seems that the position is being tracked very well, no? There's almost no noise in the position. Are these jumps noticeable while operating the robot? Or only via these plots?

The position seems well tracked, but when zooming in time you see its noisiness.
The behavior of the robot gets scary. The sounds it makes gets amplified with time, and the spikes in joint torques get linked with bumps. See in [video here](https://youtu.be/RL_IzQHRHA8)

Running the same trajectory, with PositionControl as well, but with KUKA's FRI examples (an adaptation of LBRJointSineOverlay) works like a charm. No noise at all when moving and no bumps.

If you have some time and want to see the issues yourself, maybe you could try using KUKA's positionControl to see if you experience something similar.

Thanks,
Jimmy

costashatz, 2020-07-22:
> I have modified your Java file to not use JointImpedanceControl but KUKA's JointPositionControl instead.

I see.. I have never used the pure `JointPositionControl` of KUKA.

> The sounds it makes gets amplified with time, and the spikes in joint torques get linked with bumps. See in [video here](https://youtu.be/RL_IzQHRHA8)

Yeap. I get what you mean. This is most probably because KUKA has insanely high gains for position control. In that case, you are right that you need much higher rate control (or a smarter way of sending the position commands) to not have these jumps and jerky motion.

> Running the same trajectory, with PositionControl as well, but with KUKA's FRI examples (an adaptation of LBRJointSineOverlay) works like a charm. No noise at all when moving and no bumps.

Using Java? Or via C++? Are you still controlling in 200Hz?

> If you have some time and want to see the issues yourself, maybe you could try using KUKA's positionControl to see if you experience something similar.

I do not have access to a KUKA robot at the moment. Maybe @nash169 can try out.

JimmyDaSilva, 2020-07-23:
> Using Java? Or via C++? Are you still controlling in 200Hz?

The C++ examples. But it's much simpler because it's only one thread.
In iiwa_ros, there are iiwa_driver, the controller, and a node to send the commands.

costashatz, 2020-07-23:
> The C++ examples. But it's much simpler because it's only one thread.

Hmmm... This means that ROS controllers (or our implementation of iiwa_ros driver) is adding quite some overhead. We should fix this if it comes from our code.. I'll try to create some benchmarking code..

costashatz, 2020-07-23:
Except if this is coming from the `safety limits` of ROS. Can you try changing [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa14.xacro#L37) (or [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa7.xacro#L37) depending on which robot you have) to `<xacro:property name="safety" value="false"/>` and retry? Sometimes the safety limits can interfere with the controller.

JimmyDaSilva, 2020-07-23:
Good idea.
Maybe it helped but the issue still occurs anyway...

JimmyDaSilva, 2020-07-23:
So far I have made a bunch of things that make it definitely better (smoother):
- install a PREEMPT-RT kernel, and setup like [here](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel)
- set `SCHED_FIFO` scheduling, and RT 99 priority like [here](https://github.com/frankaemika/libfranka/blob/06ad8dcf5706f00663b6fd6351734096cea9c2d0/src/control_tools.cpp#L73-L80) in `iiwa_driver` and my publisher
- isolated 2 CPUs cores from Linux general scheduler as explained [here](https://www.suse.com/support/kb/doc/?id=000017747)
- started `iiwa_driver` and my publisher on those isolated CPUs with `launch-prefix="taskset -c CPU_NB"`
- disabled the safety limits `<xacro:property name="safety" value="false"/>`

I put here my simple test package moving the 4th axis slowly with a sinus function.
[iiwa_fri_test.zip](https://github.com/epfl-lasa/iiwa_ros/files/4965937/iiwa_fri_test.zip)

There is less noise in the torques now, but it still gets crazy at some random point 
[image]

Maybe there is something in iiwa_driver that gets it out of the RT loop.

adamconkey, 2020-12-17:
@JimmyDaSilva I'm curious how your usage of iiwa_ros relates to your [lwr_hardware](https://github.com/kuka-isir/lwr_hardware) package. We've been using a variant of lwr_hardware for our LBR4 robot for years and recently got a couple iiwas. Have you used the ISIR stack for controlling the iiwa?

JimmyDaSilva, 2020-12-21:
Hi @adamconkey. My usage moved away from the `rtt-lwr` packages. 
There is no Orocos component available for the IIWA in our lab so far.

closed 2022-11-14 by costashatz