epfl-lasa/iiwa_ros #39: Using iiwa_tools requires to compile with SIMD

matthias-mayr, 2020-08-28:
I am currently exploring to use `iiwa_tools` for FK and the Jacobian in my code.

I always had crashes upon loading the controller in Gazebo when it parses the urdf to feed it to RBDyn.
It finally turned out, that the issue was, that I did not compile my new controller code with `SIMD` turned on.

I guess it would be possible to compile all the dependencies as well as `iiwa_tools` without `SIMD` and that would work as well.
But since it's the default and not directly intuitive that it is needed for derived code as well, I will leave this here as a note.

costashatz, 2020-08-28:
> It finally turned out, that the issue was, that I did not compile my new controller code with `SIMD` turned on.

The default behavior of `robot_controllers` is to compile with SIMD on. This is why `iiwa_tools` also compiles with SIMD on.

You can disable SIMD for `iiwa_tools` by passing the argument to catkin_make: `-DENABLE_SIMD=OFF`.

We should include this somewhere in the documentation..

costashatz, 2022-02-05:
This this should be fixed by #61 no?

matthias-mayr, 2022-02-06:
Should be fine. This also frees me from dragging the `SIMD` option into some projects that use `SpaceVecAlg`.

closed 2022-02-06 by matthias-mayr