epfl-lasa/iiwa_ros #40: FRI ck_compound_return_error

CKowal, 2020-10-21:
Hey everyone,

I am using FRI with the rosbridge and rossharp to publish some data to Unity while moving the robot. Recently I came across this "CK_COMPOUND_RETURN_ERROR" (see attached image for full log on the smartPAD). It states "possible connection problem" but the disconnect occurs randomly during my movements, sometimes after 30 seconds and sometimes after 10 minutes.
Without posting all of my code, does anyone have an idea where to look for the solution? What could be the cause of such a behaviour?

Best regards

[image]

costashatz, 2020-10-21:
@CKowal thanks for using the library.

This is most probably because there are many things running in your computer and the FRI driver cannot keep the control rate. A few comments/ideas that might help:

- Use one dedicated computer for the iiwa_ros driver, and run all other ROS nodes in a different computer
- Without the real time linux kernel (and the priorities well set, see [here](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel) for an example setup), it will be difficult to be able to maintain more than 200Hz for control. So make sure you set this value in the java file accordingly (see [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/java/FRIOverlay.java#L47))

Hope that I helped!

CKowal, 2020-10-21:
@costashatz 
That could be the case, I am not using a separate computer for the FRI. Thanks for the suggestions, I will try to fix it!

closed 2020-10-21 by CKowal