epfl-lasa/iiwa_ros #44: combine FRI function with smartservo

921218316, 2020-11-18:
As far as I know, Fri library has the function of torque control，and the smartservo library has the function of cartesian pose and cartesian velocity. Both libraries are connected to the manipulator via IP. Is it possible to unify the two, establish a connection with the manipulator through a network port, and use moment control (FRI) and Cartesian control (smartservo) at the same time? If so, how to change the Java code？

costashatz, 2020-12-04:
> Is it possible to unify the two, establish a connection with the manipulator through a network port, and use moment control (FRI) and Cartesian control (smartservo) at the same time?

I have no idea to be honest; I never tried to combine them. Maybe @JimmyDaSilva knows more about this..

JimmyDaSilva, 2020-12-05:
> Is it possible to unify the two, establish a connection with the manipulator through a network port, and use moment control (FRI) and Cartesian control (smartservo) at the same time? If so, how to change the Java code？

From what I have seen with both interfaces, I think having both capabilities available at runtime is not straightforward.
If hitting the robot's breaks when switching between an FRI interface and a SmartServo is an option for you I think it could be done. But definitively requires some work on the Java side.
Not sure if switching without putting the breaks is possible though...

If you want the ROS capabilities with both libraries (iiwa_ros and iiwa_stack), it is a bit more complicated.

I know the grl package at least start implementing the SmartServo and FRI libraries together in the same package: https://github.com/ahundt/grl
So I guess @ahundt is the most qualified person to answer that question I guess.

You can also have a look at the issues of @jacknlliu here: https://github.com/jacknlliu/development-issues/issues/46

Same issue has been opened here https://github.com/IFL-CAMP/iiwa_stack/issues/261

closed 2022-02-05 by costashatz