epfl-lasa/iiwa_ros #49: What is the difference between Position/Torque mode in FRIOverlay?

adamconkey, 2020-12-17:
I see in the [FRIOverlay code](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/java/FRIOverlay.java#L124-L126) that if you select torque control mode, it will zero out all the damping terms. Is that the only difference when I select torque mode on pendant instead of position mode? Because when I bring up the driver on my PC, I can select between the position and torque controller for ROS control and that will offer a corresponding command topic. I'm just wondering if there is any other change induced from the FRIOverlay on the KUKA side other than zeroing out damping terms in torque mode that I'm missing.

Thanks!

costashatz, 2020-12-17:
> Is that the only difference when I select torque mode on pendant instead of position mode?

Nope! There are several differences, the key ones being:

1. If you select torque, you will only be able to send torque commands (through the TorqueController in ROS; you have to run this controller and not the PositionController). Similarly, when selecting position control, you can only send position commands in ROS
2. I am setting the damping to zero to prevent any undesired effect from KUKA's controller. Actually I should make this a parameter just like the stiffness value. When I am controlling in torque, I set stiffness and damping to zero (one can use values a bit higher than zero for the damping)

adamconkey, 2020-12-17:
I see, so the ROS user needs to be consistent in launching the ROS controller corresponding to what's selected on the pendant, and if they mix them up and launch a different one, the robot just won't move because it's not receiving the commands it expects.

Thanks for the quick response!

closed 2020-12-17 by adamconkey