epfl-lasa/iiwa_ros #51: Compilation Issue in robot_controllers

Pranav186, 2020-12-21:
Hi guys,
I am in the last step of the documentation for the robot_controllers dependency.
I am getting the following terminal output after the command ```cmake ..```

I am pasting my terminal output here-

```
robotics@robotics-lab:~/kuka_ws/src/robot_controllers/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'eigen3'
CMake Warning (dev) at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  Policy CMP0057 is not set: Support new IN_LIST if() operator.  Run "cmake
  --help-policy CMP0057" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  IN_LIST will be interpreted as an operator when the policy is set to NEW.
  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  Policy CMP0057 is not set: Support new IN_LIST if() operator.  Run "cmake
  --help-policy CMP0057" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  IN_LIST will be interpreted as an operator when the policy is set to NEW.
  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  if given arguments:

    "_component" "IN_LIST" "_CORRADE_LIBRARY_COMPONENTS" "AND" "NOT" "_component" "IN_LIST" "_CORRADE_HEADER_ONLY_COMPONENTS"

  Unknown arguments specified
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  CMakeLists.txt:34 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/robotics/kuka_ws/src/robot_controllers/build/CMakeFiles/CMakeOutput.log".
robotics@robotics-lab:~/kuka_ws/src/robot_controllers/build$ make -j
make: *** No targets specified and no makefile found.  Stop.
robotics@robotics-lab:~/kuka_ws/src/robot_controllers/build$ sudo make install
make: *** No rule to make target 'install'.  Stop.
```

adamconkey, 2020-12-21:
I had this same problem, I fixed it by adding this line to the `robot_controllers/CMakeLists.txt` file near the top under `cmake_minimum_required`:

    cmake_policy(SET CMP0057 NEW)

Pranav186, 2020-12-21:
Thanks @adamconkey ! I am able to compile the robot_controllers package.
However, I am getting a warning and error while compiling the entire workspace.

```
CMake Warning (dev) at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  Policy CMP0057 is not set: Support new IN_LIST if() operator.  Run "cmake
  --help-policy CMP0057" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  IN_LIST will be interpreted as an operator when the policy is set to NEW.
  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  iiwa_ros/iiwa_control/CMakeLists.txt:53 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  Policy CMP0057 is not set: Support new IN_LIST if() operator.  Run "cmake
  --help-policy CMP0057" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  IN_LIST will be interpreted as an operator when the policy is set to NEW.
  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  iiwa_ros/iiwa_control/CMakeLists.txt:53 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/local/share/cmake/Corrade/FindCorrade.cmake:412 (if):
  if given arguments:

    "_component" "IN_LIST" "_CORRADE_LIBRARY_COMPONENTS" "AND" "NOT" "_component" "IN_LIST" "_CORRADE_HEADER_ONLY_COMPONENTS"

  Unknown arguments specified
Call Stack (most recent call first):
  /usr/local/share/cmake/Corrade/CorradeConfig.cmake:26 (include)
  iiwa_ros/iiwa_control/CMakeLists.txt:53 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/robotics/kuka_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/robotics/kuka_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

adamconkey, 2020-12-21:
@Pranav186 you can do the same thing for `iiwa_control/CMakeLists.txt`, I had to add it to that one also.

Pranav186, 2020-12-21:
Thanks @adamconkey ! I am able to compile the entire workspace now.

closed 2020-12-21 by Pranav186