epfl-lasa/iiwa_ros #52: Build kuka_fri with fPIC

adamconkey, 2020-12-21:
I want to build kuka_fri with the `-fPIC` flag in order to use this code in the Orocos framework (seems it's necessary otherwise the linker complains on building). I saw there was a `--shared` option for the `waf` script so I tried that, but passing that option gives me 

```
Waf: Entering directory `/home/ll4ma/source_code/iiwa_dependencies/kuka_fri/build'
[ 1/14] Compiling src/kuka/fri/LBRClient.cpp
[ 2/14] Compiling src/kuka/fri/MonitoringMessageDecoder.cpp
[ 3/14] Compiling src/kuka/fri/UdpConnection.cpp
[ 4/14] Compiling src/kuka/fri/CommandMessageEncoder.cpp
[ 5/14] Compiling src/kuka/fri/LBRCommand.cpp
[ 6/14] Compiling src/kuka/pb/pb_decode.c
[ 7/14] Compiling src/kuka/fri/FRIMessages.pb.c
[ 8/14] Compiling src/kuka/pb/pb_frimessages_callbacks.c
[ 9/14] Compiling src/kuka/fri/ClientApplication.cpp
[10/14] Compiling src/kuka/pb/pb_encode.c
[11/14] Compiling src/kuka/fri/LBRState.cpp
[12/14] Compiling src/example.cpp
[13/14] Linking build/libkuka_fri.so
/usr/bin/ld: src/kuka/pb/pb_encode.c.1.o: relocation R_X86_64_PC32 against symbol `pb_encode_tag' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
```

I tried manually adding the `-fPIC` flag in the configure function in `wscript` but I can't seem to be able to get around this error. Any ideas? I'm not at all familiar with `waf` so I'm a little clueless here. I might just write a Makefile unless there's a reason why cmake isn't being used to build this package.

costashatz, 2021-03-10:
@adamconkey did you manage to solve this? (really sorry for the super late answer!)

adamconkey, 2021-03-16:
Yes I'll close the issue. I ended up just writing a makefile so I could build it with cmake instead.

closed 2021-03-16 by adamconkey