epfl-lasa/iiwa_ros #55: Compilation with iiwa_service

Lejou, 2021-03-26:
I have already install all the dependencies that the project needs, and when I "catkin_make" my ROS workspace, I have the error with iiwa_service, like this:

[image]

How can I figure out this problem? What happened to this? Could anyone give me a hand?

matthias-mayr, 2021-03-26:
It's a linker error since it can not link to one of the dependencies, `rbd`. The first thing that you can check  that you correctly compiled and installed the dependencies.

costashatz, 2021-03-26:
Did you install the non-ROS dependencies? See [here](https://github.com/epfl-lasa/iiwa_ros#dependencies) for instructions..

Lejou, 2021-03-27:
I am pleased to tell you that what u guys told me is right. Some dependencies are not installed in the right way. I install them again, and the compilation is complete. Appreciate! Thank you very much!!!!

closed 2021-03-27 by Lejou