epfl-lasa/iiwa_ros #56: How to use REALLY Pure Postion mode with FRI?

longwoo, 2021-04-10:
Thank you for your great job !

I had already use your driver to move my iiwa7 .

Now, I can use moveit in Rivz to drag the robot and the real iiwa can follow the drag trajectory.

The question is : the robot is not actually in Pure Postion Mode, even the stiffness is set to high value, and the robot was not accurate acchieve the goal which effected by the gravity.

the JAVA code in robot side is 

`
        // configure and start FRI session
        FRIConfiguration friConfiguration = FRIConfiguration.createRemoteConfiguration(_lbr, _clientName);
        // for torque mode, there has to be a command value at least all 5ms
        friConfiguration.setSendPeriodMilliSec(5);
        friConfiguration.setReceiveMultiplier(1);

        getLogger().info("Creating FRI connection to " + friConfiguration.getHostName());
        getLogger().info("SendPeriod: " + friConfiguration.getSendPeriodMilliSec() + "ms |"
                + " ReceiveMultiplier: " + friConfiguration.getReceiveMultiplier());

        FRISession friSession = new FRISession(friConfiguration);

        // wait until FRI session is ready to switch to command mode
        try
        {
            friSession.await(10, TimeUnit.SECONDS);
        }
        catch (final TimeoutException e)
        {
            getLogger().error(e.getLocalizedMessage());
            friSession.close();
            return;
        }

        getLogger().info("FRI connection established.");

        ClientCommandMode mode = ClientCommandMode.POSITION;
        FRIJointOverlay jointOverlay = new FRIJointOverlay(friSession, mode);
        double stiffness = 500.;

        // start PositionHold with overlay
        JointImpedanceControlMode ctrMode = new JointImpedanceControlMode(stiffness, stiffness, stiffness, stiffness, stiffness, stiffness, stiffness);

        try {
            PositionHold posHold = new PositionHold(ctrMode, -1, TimeUnit.SECONDS);
            getLogger().info("Robot is ready for ROS control.");
            _lbr.move(posHold.addMotionOverlay(jointOverlay));
        }
        catch(final CommandInvalidException e) {
            getLogger().error("ROS has been disconnected.");
        }

        // done
        friSession.close();

`

How can I just use pure postion mode to do position trajectory task?

costashatz, 2021-04-11:
Hello,

Not sure what you mean by "pure position control". But RViz can create delays in sending commands to the robot. In any case, try the following:

- Increase the stiffness to 1000
- You can put a smaller value (than the default) for damping (see [here](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/java/FRIOverlay.java#L126)): a value like 0.5 should be ok
- Check the delays created by RViz

With all the above I was able to control the robot in mm accuracy (see [ref](https://infoscience.epfl.ch/record/275435)). BUT I would vote that the issue comes from RViz and not the driver: a stiffness of 1000 can make the robot quite dangerous (I really needed mm accuracy in end-effector; for all other movements a stiffness of 200-400 was always more than enough)! Be careful!

Best,
Konstantinos

JimmyDaSilva, 2021-04-12:
Hi @longwoo,

I understand that you want to use KUKA's internal "Position" controller instead of the "Joint/Cartesian Impedance" controllers.
This is quite simple!

On your Java application,
 **_remove_**:
```      
JointImpedanceControlMode ctrMode = new JointImpedanceControlMode(stiffness, stiffness, stiffness, stiffness, stiffness, stiffness, stiffness);
PositionHold posHold = new PositionHold(ctrMode, -1, TimeUnit.SECONDS);
```
 **_add_**:
```
PositionControlMode pos_controller = new PositionControlMode();
```

@costashatz The position controller has the strongest gains possible, and an integral correction.
Though it allows less cooperation but is more precise.

@longwoo Use it with caution! Actually, I don't remember trying this with the iiwa_ros package.

longwoo, 2021-04-12:
> Hello,
>
> Not sure what you mean by "pure position control". But RViz can create delays in sending commands to the robot. In any case, try the following:
>
>     * Increase the stiffness to 1000
>
>     * [...]

Hi, 

Thank you. I solved this by changing Java code in iiwa side

```
        ClientCommandMode mode = ClientCommandMode.POSITION;
        FRIJointOverlay jointOverlay = new FRIJointOverlay(friSession, mode);
        double stiffness = 500.;

        // start PositionHold with overlay
        PositionControlMode ctrMode = new PositionControlMode();

        try {
            PositionHold posHold = new PositionHold(ctrMode, -1, TimeUnit.SECONDS);
            getLogger().info("Robot is ready for ROS control.");
            _lbr.move(posHold.addMotionOverlay(jointOverlay));
        }
        catch(final CommandInvalidException e) {
            getLogger().error("ROS has been disconnected.");
        }
```

The reason why I need a "pure position control" before is that force control was not needed in my experiment now.  

What I want is just doing some trajectory following in real time.

As you may know, FRI is not very straightfoward  in iiwa side since no joint position or joint torque is obviously assigned.  I thought `PositionHold` must be used together with `JointImpedanceControlMode`, when using FRI in iiwa side, which means I have to use "force control " to simulate "position control".

Now it was solved.

longwoo, 2021-04-12:
> Hi @longwoo,
> 
> I understand that you want to use KUKA's internal "Position" controller instead of the "Joint/Cartesian Impedance" controllers.
> This is quite simple!
> 
> On your Java application,
> **_remove_**:
> 
> ```
> JointImpedanceControlMode ctrMode = new JointImpedanceControlMode(stiffness, stiffness, stiffness, stiffness, stiffness, stiffness, stiffness);
> ```
> 
> **_add_**:
> 
> ```
> PositionControlMode ctrMode = new PositionControlMode();
> ```
> 
> @costashatz The position controller has the strongest gains possible, and an integral correction.
> So it allows for less cooperation but is more precise.
> 
> @longwoo Use it with caution! Actually, I don't remember trying this with the iiwa_ros package.

@JimmyDaSilva  Thank you for your help. I had just verified this method hours ago. It can work with your iiwa_ros package.     

Cheer ^.^

closed 2021-04-13 by costashatz

priyampnchl, 2023-10-06:
@longwoo is it possible to take a look at your code for commanding the robot in Position Control mode through ROS?