epfl-lasa/iiwa_ros #57: how to fix the FK gap between urdf model and real robot?

longwoo, 2021-04-16:
Hi,
With the help of your work , I can communicate iiwa7 with FRI. 

As we know, Fri only support joint relate information.

One way to get end effector pos is doing a forward kinematics (FK) with URDF model.

However, the URDF model in this ros package is a idea model and is different from the real iwwa7 model, which has been calibrated.

As a result, the value solved by [iiwa_tools](https://github.com/epfl-lasa/iiwa_ros/tree/master/iiwa_tools) is different from it was solved by real iwwa7.

For example , 
the FK  `x, y, z` result solved by iiwa_tools is `0.2270, 0.3588, 0.6772`,
the FK `x, y, z` result solved by real iiwa is `0.23197, 0.36651, 0.65290`

So I am wondering how you fix the FK gap between  iiwa_tools and real iiwa?

costashatz, 2021-04-16:
> However, the URDF model in this ros package is a idea model and is different from the real iwwa7 model, which has been calibrated and more accurate.

Are you sure about this? I was mainly using the iiwa14 model, and I checked it many times and it was perfect.

It might be that you have changed the reference frame? I remember at some point we had put motion capture markers and the estimated base was different from the URDF. Is this what you are doing? If this is the case, I have a few ideas..

matthias-mayr, 2021-04-16:
I have once done a small test where I probed different positions with an iiwa7 on the table you see in [this video](https://www.youtube.com/watch?v=KOeE55GHQhY&t=10s) that has a known height of 60cm. The z-values (height) were only off by a couple of millimeters.
Our robot is mounted in tilted, so chances would be that errors in the URDF would result in offsets in real world coordinates.
I have never queried the KUKA FK though - in our mounting configuration this is not a useful frame.

closed 2022-02-05 by costashatz