epfl-lasa/iiwa_ros #63: safety error in safety configuration of sunrise workbench and in smartpad

mf093087, 2021-08-22:
I am following the tutorials to use the KUKA iiwa package. 
1. In the 'Sunrise Robot Application upload' section, I need to set the safety configurations. In my sunrise, I can only set 3 axises not 7 axises.
[image]

2. When I modify my sunrise file without the safety configuration, I synchronize the project to smartPAD. In my smart pad, the light for the safety is always red. 
I met the error 
_Safety control
position sensor not referenced(K1:A1,A2,A3,A4,A5,A6,A7).
Emergency stop external is active.
protective stop is activated_
[image]

mf093087, 2021-08-23:
problem solved with deactivate the emergency stop external and protective stop in KUKA sunrise workbench

costashatz, 2021-08-23:
@mf093087 good to know! Closing the issue.

closed 2021-08-23 by costashatz