epfl-lasa/iiwa_ros #65: Connection issues, timeout before FRI connection quality reached good or excellent.

mf093087, 2021-08-24:
I set up the robot and met this problem on the smart pad. 

_timeout before FRI connection quality reached good or excellent._
[image]

I tried the solution, which is provided by  921218316. 
https://github.com/epfl-lasa/iiwa_ros/issues/17

But it still doesn't work, then I tried this solution, it still didn't work. Could you please help me? I assume it is because I am using ubuntu virtual machine on the same computer of the windows laptop. Should I use two separe computers?

mf093087, 2021-09-05:
When I switch to dual system, it works. I am not sure why it doesn't work in virtual machine, can anybody help me?

costashatz, 2021-09-22:
> I am not sure why it doesn't work in virtual machine, can anybody help me?

My best guess is that Virtual Machine does not have enough CPU power to keep the connection loop at 1KHz/500Hz.

mf093087, 2021-09-23:
I tried dual system, it worked. But I think you should give an instruction about how to use the library on virtual machine. That would be very helpful.

costashatz, 2021-09-23:
> I tried dual system, it worked. But I think you should give an instruction about how to use the library on virtual machine. That would be very helpful.

As I said in the comment above: `I do not think that the Virtual Machine have enough CPU power to keep the connection loop at 1KHz/500Hz.` Maybe if you give it more vCPUs, but still you would need priority to serial ports etc. Not worth the work.

Closing this issue..

closed 2021-09-23 by costashatz

mf093087, 2021-09-23:
I don't think it has something to do with the CPU power, in my case, I already gave enough CPU power. I think something is not correct with the IP settings. I already assigned a static IP address to the virtual machine. But because I am not so familiar with the virtual machine IP setting, so I can't still connect to the robot with virtual machine. If an instruction about how to set the IP address and ping the robot in virtual machine would be very nice. However, if it is not worth the work, I might try to find solutions from somewhere else. 

Many thanks for your help, I think your library is very nice, this help me a lot.

costashatz, 2021-09-23:
> I don't think it has something to do with the CPU power, in my case, I already gave enough CPU power. I think something is not correct with the IP settings. I already assigned a static IP address to [...]

I am pretty sure that even if you fix the IP issues (not sure IP setting is different than other PCs; I have barely used VMs --- my guess would be that you are not using the ethernet ports correctly), this will not be able to keep up the 500Hz/1KHz. We had big machine running native linux that if we didn't set high performance profile and we didn't give priority to hardware devices, they were not able to keep up the communication. So I really doubt that using a VM to run the iiwa_ros controller is a good option.

mf093087, 2021-09-23:
> I am pretty sure that even if you fix the IP issues (not sure IP setting is different than other PCs; I have barely used VMs --- my guess would be that you are not using the ethernet ports [...]

Hi Costas, I think this makes sense, now I think I will keep on using a dual system, not VM. Because before I was trying to implement my approach on a dual system to VM. Now it seems to be not a good option...