epfl-lasa/iiwa_ros #66: Question about Sunrise Robot Application upload

zxl5344, 2021-08-31:
I am a beginner of using iiwa robot. 

In repo, It mentioned: "Create a new Sunrise project with Sunrise Workbench". Where do I create this Sunrise project? On "a Windows laptop is connected on the X66 Ethernet port" or I can directly create on KUKA Sunrise Cabinet?  

If I have to use a laptop, how do I get Sunrise Workbench? 

Many thanks

costashatz, 2021-09-01:
> "a Windows laptop is connected on the X66 Ethernet port"

If I understand well your question. This is where you should create the Sunrise Project.

> If I have to use a laptop, how do I get Sunrise Workbench?

KUKA should have given you this.

zxl5344, 2021-09-02:
Thank you very much!

closed 2021-09-02 by zxl5344