epfl-lasa/iiwa_ros #69: [iiwa/iiwa_service-2] process has died [pid 13653, exit code 127, cmd /home/fan/catkin_ws/devel/lib/iiwa_tools/iiwa_service __name:=iiwa_service __log:=/home/fan/.ros/log/9b6ea0cc-0e32-11ec-bd82-4851c55e749b/iiwa-iiwa_service-2.log]. log file: /home/fan/.ros/log/9b6ea0cc-0e32-11ec-bd82-4851c55e749b/iiwa-iiwa_service-2*.log

mf093087, 2021-09-05:
I successfully connect with the robot via FRI. But when I run 'roslaunch iiwa_driver iiwa_bringup.launch'. I met this error. 
/home/fan/Pictures/Screenshot from 2021-09-05 11-19-38.png

And when I run  'roslaunch iiwa_gazebo iiwa_gazebo.launch', I met this error.
/home/fan/Pictures/Screenshot from 2021-09-05 11-23-45.png

Maybe I haven't installed the dependencies correctly. Could you please help me?

costashatz, 2021-09-22:
> 'roslaunch iiwa_driver iiwa_bringup.launch'

This is intended to work only with the physical robot. You must ensure that the FRI app is running on the real KUKA and a connection is feasible.

> And when I run 'roslaunch iiwa_gazebo iiwa_gazebo.launch'

`roslaunch iiwa_gazebo iiwa_gazebo.launch` should be ran without `iiwa_bringup.launch`.

zxl5344, 2021-12-03:
I have the same issues when I run command  'roslaunch iiwa_gazebo iiwa_gazebo.launch'. Other function like moveit is working. But there is no iiwa_service. Is anyone know how to solve it?

Here is my log:
... logging to /home/zxl5344/.ros/log/12d921ec-546e-11ec-bb30-45a21bc009e3/roslaunch-E5-ME-L-KAFI11-6075.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://E5-ME-L-KAFI11:35327/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /iiwa/CustomControllers/controllers/LinearDS/params: [1.0, 1.0, 1.0]
 * /iiwa/CustomControllers/controllers/LinearDS/type: LinearDSController
 * /iiwa/CustomControllers/controllers/PD/input: Orientation
 * /iiwa/CustomControllers/controllers/PD/output: Torque
 * /iiwa/CustomControllers/controllers/PD/params: [20.0, 2.0]
 * /iiwa/CustomControllers/controllers/PD/type: PidController
 * /iiwa/CustomControllers/controllers/PassiveDS/params: [200, 200]
 * /iiwa/CustomControllers/controllers/PassiveDS/type: PassiveDSController
 * /iiwa/CustomControllers/joints: ['iiwa_joint_1', ...
 * /iiwa/CustomControllers/params/end_effector: iiwa_link_ee
 * /iiwa/CustomControllers/params/null_space/Kd: 0.1
 * /iiwa/CustomControllers/params/null_space/Kp: 20.0
 * /iiwa/CustomControllers/params/null_space/joints: [0.04475269104532...
 * /iiwa/CustomControllers/params/null_space/max_torque: 10.0
 * /iiwa/CustomControllers/params/space: task
 * /iiwa/CustomControllers/structure/CascadeCtrl: ['LinearDS', 'Pas...
 * /iiwa/CustomControllers/type: iiwa_control/Cust...
 * /iiwa/PositionController/joints: ['iiwa_joint_1', ...
 * /iiwa/PositionController/type: position_controll...
 * /iiwa/PositionTorqueController/iiwa_joint_1/pid/d: 80.0
 * /iiwa/PositionTorqueController/iiwa_joint_1/pid/i: 100
 * /iiwa/PositionTorqueController/iiwa_joint_1/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_1/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_1/pid/p: 800.0
 * /iiwa/PositionTorqueController/iiwa_joint_2/pid/d: 100.0
 * /iiwa/PositionTorqueController/iiwa_joint_2/pid/i: 1000
 * /iiwa/PositionTorqueController/iiwa_joint_2/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_2/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_2/pid/p: 800.0
 * /iiwa/PositionTorqueController/iiwa_joint_3/pid/d: 80.0
 * /iiwa/PositionTorqueController/iiwa_joint_3/pid/i: 10
 * /iiwa/PositionTorqueController/iiwa_joint_3/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_3/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_3/pid/p: 800.0
 * /iiwa/PositionTorqueController/iiwa_joint_4/pid/d: 80.0
 * /iiwa/PositionTorqueController/iiwa_joint_4/pid/i: 10
 * /iiwa/PositionTorqueController/iiwa_joint_4/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_4/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_4/pid/p: 800.0
 * /iiwa/PositionTorqueController/iiwa_joint_5/pid/d: 10.0
 * /iiwa/PositionTorqueController/iiwa_joint_5/pid/i: 5
 * /iiwa/PositionTorqueController/iiwa_joint_5/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_5/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_5/pid/p: 300.0
 * /iiwa/PositionTorqueController/iiwa_joint_6/pid/d: 1.0
 * /iiwa/PositionTorqueController/iiwa_joint_6/pid/i: 1
 * /iiwa/PositionTorqueController/iiwa_joint_6/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_6/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_6/pid/p: 50.0
 * /iiwa/PositionTorqueController/iiwa_joint_7/pid/d: 1.0
 * /iiwa/PositionTorqueController/iiwa_joint_7/pid/i: 0.1
 * /iiwa/PositionTorqueController/iiwa_joint_7/pid/i_clamp_max: 10000
 * /iiwa/PositionTorqueController/iiwa_joint_7/pid/i_clamp_min: -10000
 * /iiwa/PositionTorqueController/iiwa_joint_7/pid/p: 10.0
[... 125 lines cut ...]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Invalid partition name [E5-ME-L-KAFI11:zxl5344@AD.PSU.EDU]
Topic [/iiwa/joint_cmd] is not valid.
Service [/iiwa/joint_cmd_req] is not valid.
[INFO] [1638559232.042226, 0.368000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1638559232.113604582, 0.368000000]: Loading gazebo_ros_control plugin
[ INFO] [1638559232.113688934, 0.368000000]: Starting gazebo_ros_control plugin in namespace: /iiwa
[ INFO] [1638559232.114030362, 0.368000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1638559232.235252053, 0.368000000]: Loaded gazebo_ros_control.
[INFO] [1638559232.390294, 0.523000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1638559232.392689, 0.525000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[... cut ...] controller: /iiwa/TorqueController
[INFO] [1638559232.406171, 0.539000]: Loading controller: /iiwa/joint_state_controller
[INFO] [1638559232.414103, 0.547000]: Controller Spawner: Loaded controllers: /iiwa/TorqueController, /iiwa/joint_state_controller
[INFO] [1638559232.418280, 0.551000]: Started controllers: /iiwa/TorqueController, /iiwa/joint_state_controller
[urdf_spawner-5] process has finished cleanly
log file: /home/zxl5344/.ros/log/12d921ec-546e-11ec-bb30-45a21bc009e3/urdf_spawner-5*.log

costashatz, 2022-02-05:
> /home/zxl5344/IROS2022/devel/lib/iiwa_tools/iiwa_service: error while loading shared libraries: libRBDyn.so.1: cannot open shared object file: No such file or directory

You do not have installed correctly the dependencies for iiwa_service. Please follow the installation instructions..

> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Service [/physics/info/plugin] is not valid.
> Service [/scene_info] is not valid.
> Service [/shadow_caster_material_name] is not valid.
> Service [/shadow_caster_render_back_faces] is not valid.
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Service [/introspection/tnuhez/filter_new] is not valid.
> Service [/introspection/tnuhez/filter_update] is not valid.
> Service [/introspection/tnuhez/filter_remove] is not valid.
> Service [/introspection/tnuhez/items] is not valid.
> Topic [/introspection/tnuhez/items_update] is not valid.
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]
> Invalid partition name [E5-ME-L-KAFI11:[zxl5344@AD.PSU.EDU](mailto:zxl5344@AD.PSU.EDU)]

Overall this seems that you are either mixing ROS versions or your ROS is not installed correctly...

closed 2022-02-05 by costashatz