epfl-lasa/iiwa_ros #70: Motion paused during PositionControl

zxl5344, 2021-09-14:
Hi, all

I am using PositionController from repo. I wrote a python code to let robot move from joint angle [0, 0, 0, 0, 0, 0, 0] to joint angle [0.5, 0.2, 0.5, 0.3, 0.5, -0.5, 0.5]. My code works fine on Gazebo. But when I tried to run same code on iiwa, robot paused. The smart pad shows me: 
[image]
Why does this happen and how do I solve it or avoid it?

Thank you

costashatz, 2021-09-16:
Hey,

This "smells" like a connection issue. Most probably you do not adhere to (at least) 500Hz. Another possible issue is a bad controller. If you attempt to give very high-velocities/torques, the robot will complain and stop.

zxl5344, 2021-09-17:
Hi costashatz,

Thanks for your response.
I tried to use Torque Controller and still have the same issues. my frequency is 1000Hz. I set up a limit for max torque at [10, 10, 3., 2., 2., 2., 1.]. I am using a PD controller. My P value are: [800, 800, 800, 800, 300, 50, 10] and D is [1000, 1000, 200, 100, 5, 1, 0.1]. I run my code at stiffness = 500.

Here is my code:

`#!/usr/bin/env python
"""
This node use PD contoller to move robot back to zero position
SUBSCRIBERS:
	- /iiwa/joint_states (JointState)
PUBLISHERS:
	- /iiwa/PositionController/command (Float64MultiArray)
SERVICES:
	- /iiwa/iiwa_ik_server
	- /iiwa/iiwa_fk_server

"""

import rospy
from std_msgs.msg import Float64MultiArray, MultiArrayLayout, MultiArrayDimension
from sensor_msgs.msg import JointState
from iiwa_tools.srv import GetIK, GetIKRequest, GetFK
from geometry_msgs.msg import Pose
import numpy as np

class Move_Torque:
	"""
	this class move robot to a certain joint state with torque input use PD controller
	"""
	
	def __init__(self, _target):
		rospy.loginfo("Robot starts moving")
		self.DT=1/1000							# change frequence to 1000hz
		
		# setup state variables
		self.rob_state=JointState()
		
		# initial storing
		self.measurement = np.zeros(21)
		
		# Torque saturation 
		self.sat = [10, 10, 3., 2., 2., 2., 1.]			# stiffness 500
		
		# initial error of pos
		self.error = [0, 0, 0, 0, 0, 0, 0]
		
		# initial error of velocity
		self.errorV = [0, 0, 0, 0, 0, 0, 0]
		
		# set up P gain
		self.P = [800, 800, 800, 800, 300, 50, 10]
		
		# set up D gain
		self.D = [1000, 1000, 200, 100, 5, 1, 0.1]
		
		# setup publishers, subscribers, timers:
		self.move_pub=rospy.Publisher("/iiwa/TorqueController/command", Float64MultiArray, queue_size=10)
		self.joint_sub=rospy.Subscriber("/iiwa/joint_states", JointState, self.update_state, queue_size=10)
		rospy.wait_for_message("/iiwa/joint_states", JointState)
		self.sim_timer = rospy.Timer(rospy.Duration(self.DT),self.moving)
		
		# setup goal joint state
		self.goal = Float64MultiArray()
		self.goal.data = _target
		
		# setup Torque
		self.Torque = Float64MultiArray()
		self.Torque.data = [0, 0, 0, 0, 0, 0, 0]
[... 119 lines cut ...]
		rospy.loginfo("pose1")
		jointstate1 = [1, 1, 1, 1, 0.5, 0.5, 0.5]
		test.update_target(jointstate1)
		rospy.sleep(10.)
			
		# read current position in world frame
  		test.currentposition()
				
		rospy.loginfo("pose2")
		jointstate2 = [-1, -1, -1, -1, -0.5, -0.5, -0.5]
		test.update_target(jointstate2) 
		rospy.sleep(10.)
			
		# read current position in world frame
		test.currentposition()
		  
	except rospy.ROSInterruptException: 
		pass
	
	rospy.loginfo("task complete")

	rospy.spin()

if __name__=='__main__':
    main()`

costashatz, 2021-09-17:
> my frequency is 1000Hz

Are you sure that your communication is always at 1000Hz? I am not talking about what you select in iiwa_ros, but the real physical communication.

> I set up a limit for max torque at [10, 10, 3., 2., 2., 2., 1.]. I am using a PD controller. My P value are: [800, 800, 800, 800, 300, 50, 10] and D is [1000, 1000, 200, 100, 5, 1, 0.1]. I run my code at stiffness = 500.

I cannot really tell anything with these values, although the D values look big.

Overall you should be careful with:

- Ensuring the communicate rate at the one you set in KUKA Sunrise Java app
- Sending "smooth" commands (i.e., commands that will not make the robot change in 1-2 timesteps the whole trajectory)

Not sure if I can help more than that without any measurement or analysis from your side.

zxl5344, 2021-10-05:
Thank you for your help @costashatz 
Could you give me some suggestions about what should I measure to debug this issues. I am a beginner of robotics so your suggestion are very valuable to me.

zxl5344, 2021-11-05:
I rerun my code this week and my problem magically disappeared. Thank you for your help

closed 2021-11-05 by zxl5344