epfl-lasa/iiwa_ros #71: catkin_make problem

ismaisma, 2021-11-15:
Hi; I have a problem on compiling my workspace with ROS Melodic 
[ 95%] Linking CXX executable /home/akliisma/catkin_ws/devel/lib/iiwa_driver/iiwa_driver
CMakeFiles/iiwa_service.dir/src/iiwa_service.cpp.o : Dans la fonction « mc_rbdyn_urdf::URDFParserResult::URDFParserResult() » :
iiwa_service.cpp:(.text._ZN13mc_rbdyn_urdf16URDFParserResultC2Ev[_ZN13mc_rbdyn_urdf16URDFParserResultC5Ev]+0x24) : référence indéfinie vers « rbd::MultiBody::MultiBody() »
iiwa_service.cpp:(.text._ZN13mc_rbdyn_urdf16URDFParserResultC2Ev[_ZN13mc_rbdyn_urdf16URDFParserResultC5Ev]+0x73) : référence indéfinie vers « rbd::MultiBodyGraph::MultiBodyGraph(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) »
CMakeFiles/iiwa_service.dir/src/iiwa_service.cpp.o : Dans la fonction « mc_rbdyn_urdf::URDFParserResult::~URDFParserResult() » :
iiwa_service.cpp:(.text._ZN13mc_rbdyn_urdf16URDFParserResultD2Ev[_ZN13mc_rbdyn_urdf16URDFParserResultD5Ev]+0x50) : référence indéfinie vers « rbd::MultiBodyGraph::~MultiBodyGraph() »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::MultiBodyGraph::MultiBodyGraph(rbd::MultiBodyGraph const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::ForwardDynamics::computeC(rbd::MultiBody const&, rbd::MultiBodyConfig const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::forwardVelocity(rbd::MultiBody const&, rbd::MultiBodyConfig&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::MultiBodyConfig::zero(rbd::MultiBody const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::Jacobian::jacobianDot(rbd::MultiBody const&, rbd::MultiBodyConfig const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::Jacobian::Jacobian(rbd::MultiBody const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::MultiBodyGraph::operator=(rbd::MultiBodyGraph const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::InverseKinematics::inverseKinematics(rbd::MultiBody const&, rbd::MultiBodyConfig&, sva::PTransform<double> const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::Jacobian::jacobian(rbd::MultiBody const&, rbd::MultiBodyConfig const&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::InverseKinematics::InverseKinematics(rbd::MultiBody const&, int) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « mc_rbdyn_urdf::rbdyn_from_urdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::forwardKinematics(rbd::MultiBody const&, rbd::MultiBodyConfig&) »
/home/akliisma/catkin_ws/devel/lib/libiiwa_tools.so : référence indéfinie vers « rbd::ForwardDynamics::ForwardDynamics(rbd::MultiBody const&) »
collect2: error: ld returned 1 exit status
iiwa_ros/iiwa_tools/CMakeFiles/iiwa_service.dir/build.make:157: recipe for target '/home/akliisma/catkin_ws/devel/lib/iiwa_tools/iiwa_service' failed
make[2]: *** [/home/akliisma/catkin_ws/devel/lib/iiwa_tools/iiwa_service] Error 1
CMakeFiles/Makefile2:11349: recipe for target 'iiwa_ros/iiwa_tools/CMakeFiles/iiwa_service.dir/all' failed
make[1]: *** [iiwa_ros/iiwa_tools/CMakeFiles/iiwa_service.dir/all] Error 2
make[1]: *** Attente des tâches non terminées....
[ 95%] Built target iiwa_driver
[ 95%] Built target CustomEffortController
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
 

Thanks 

P.S: sorry, the comments are in french

matthias-mayr, 2021-11-15:
You're having linker errors. Can you double-check that the dependencies installed correctly? Especially `RBDyn` and `mc_rbdyn_urdf`.

ismaisma, 2021-11-16:
new Problem : 

In file included from /usr/local/include/RBDyn/IK.h:12:0,
                 from /home/akliisma/catkin_ws/src/iiwa_ros/iiwa_tools/include/iiwa_tools/iiwa_tools.h:33,
                 from /home/akliisma/catkin_ws/src/iiwa_ros/iiwa_tools/src/iiwa_tools.cpp:25:
/usr/local/include/SpaceVecAlg/SpaceVecAlg:12:10: fatal error: Eigen/Core: Aucun fichier ou dossier de ce type
 #include <Eigen/Core>
          ^~~~~~~~~~~~
compilation terminated.
iiwa_ros/iiwa_tools/CMakeFiles/iiwa_tools.dir/build.make:62: recipe for target 'iiwa_ros/iiwa_tools/CMakeFiles/iiwa_tools.dir/src/iiwa_tools.cpp.o' failed
make[2]: *** [iiwa_ros/iiwa_tools/CMakeFiles/iiwa_tools.dir/src/iiwa_tools.cpp.o] Error 1
CMakeFiles/Makefile2:11247: recipe for target 'iiwa_ros/iiwa_tools/CMakeFiles/iiwa_tools.dir/all' failed
make[1]: *** [iiwa_ros/iiwa_tools/CMakeFiles/iiwa_tools.dir/all] Error 2
make[1]: *** Attente des tâches non terminées....
[  1%] Built target _opencv_apps_generate_messages_check_deps_RectArray
[  1%] Built target _iiwa_tools_generate_messages_check_deps_GetFK
[  1%] Built target _iiwa_tools_generate_messages_check_deps_GetJacobian
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

matthias-mayr, 2021-11-16:
Looks like ´Eigen´ is not installed. Usually it comes the latest when installing ROS.

btw, Github supports markdown for formatting. That makes code and terminal output usually easier to read: https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

closed 2022-02-05 by costashatz