epfl-lasa/iiwa_ros #73: Can't connect MoveIt! and gazebo

mf093087, 2021-12-05:
I want to use moveit and gazebo to do co-simulation. I plan the path in RVIZ with moveit, and then if gazebo is connected, the robot in gazebo will also move. 

I use following command to open moveit and gazebo in this repository. 
- roslaunch iiwa_gazebo iiwa_gazebo.launch
- roslaunch iiwa_moveit demo.launch driver:=true

[image]

The gazebo environment will not move. I think it is not connected properly. I saw this issue 
https://github.com/epfl-lasa/iiwa_ros/issues/33

I tried the following command. 

**_In order to make the setup work for you, the only thing you need to change is this line to contain

<arg name="controller" default="PositionTrajectoryController"/>
as an alternative you can also run

roslaunch iiwa_gazebo iiwa_gazebo.launch controller:=PositionTrajectoryController_**

But it still doesn't work, can anybody help me?

costashatz, 2022-02-05:
> * oslaunch iiwa_moveit demo.launch driver:=true

Here you are launching the MoveIt! for the real robot (`driver:=true` does that). You need to run this with `driver:=false`.

closed 2022-02-05 by costashatz

mf093087, 2022-02-08:
> > * oslaunch iiwa_moveit demo.launch driver:=true
> 
> Here you are launching the MoveIt! for the real robot (`driver:=true` does that). You need to run this with `driver:=false`.

Thanks a lot, I solved this problem. I need to change the controller from Torque controller to positiontrajectory controller.