epfl-lasa/iiwa_ros #74: ROS has been disconnected

zxl5344, 2021-12-07:
Hi, 

I am using a torque controller for my control research. iiwa always disconnects with PC without any warning or error. The smartPad only show  'ROS has been disconnected'. 
[image]
However, when I switch to another application on smartPad and run it, I get this error

[image]

Does anyone have clue about the cause of my problem? Thank you!

costashatz, 2022-02-05:
> 'ROS has been disconnected'.

This most probably means that you haven't configured the network setup correctly. Please check the README on how to do it properly.

closed 2022-02-07 by zxl5344

LJXASD, 2022-05-24:
Hello, I have also encountered this problem, "ros has been disconnected".
Have you solved this problem?
Thanks a lot! 

> Hi,
>
> I am using a torque controller for my control research. iiwa always disconnects with PC without any warning or error. The smartPad only show 'ROS has been disconnected'. [image] However, when I [...]