epfl-lasa/iiwa_ros #78: Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/user/catkin2iiwa/src/iiwa_description/launch/iiwaiiwa14_upload.launch' The traceback for the exception was written to the log file

mf093087, 2022-01-22:
user@ubuntu:~/catkin2iiwa$ roslaunch iiwa_moveit moveit_planning_execution.launch
... logging to /home/user/.ros/log/f9a8cdae-7bc7-11ec-9cf0-000c2940e86e/roslaunch-ubuntu-30700.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/user/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

RLException: while processing /home/user/catkin2iiwa/src/iiwa_gazebo/launch/iiwa_gazebo.launch:
while processing /home/user/catkin2iiwa/src/iiwa_driver/launch/iiwa_setup.launch:
while processing /home/user/catkin2iiwa/src/iiwa_description/launch/iiwaiiwa14_upload.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/user/catkin2iiwa/src/iiwa_description/launch/iiwaiiwa14_upload.launch'
The traceback for the exception was written to the log file

[image]

costashatz, 2022-02-05:
Did you solve this? This is surely a problem in your configuration: `iiwaiiwa14_upload.launch` would never be called with the default launch files..

mf093087, 2022-02-08:
Yes, I solved this, I changed the name, then it works..