epfl-lasa/iiwa_ros #79: Can we use moveit setup assistant to edit the file? unable to locate the URDF file in package. Expected File: /home/user/catkin_iiwa/src/iiwa_ros/iiwa_description/urdf/iiwa14.urdf

mf093087, 2022-02-03:
I want to use moveit setup assistant to edit the configuration file. However, it doesn't work for me. Can anybody help me?

unable to locate the URDF file in package. Expected File: /home/user/catkin_iiwa/src/iiwa_ros/iiwa_description/urdf/iiwa14.urdf

[image]

costashatz, 2022-02-03:
Hello. Thanks for using the library.

I think you need to generate the file (iiwa14.urdf) using the [xacro file](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_description/urdf/iiwa14.urdf.xacro) provided in the `iiwa_description` package. @ymollard did the MoveIt! integration though, so he might know better.

mf093087, 2022-02-04:
thanks a lot!

closed 2023-04-28 by costashatz