epfl-lasa/iiwa_ros #82: Timeout before FRI connection is established

cmheunis, 2022-03-24:
**Context:** We have followed the steps as per the Git instructions. Establishing the KONI connection works fine and we do receive a return on pining the IP address 172.31.1.41. We have Activated AUT mode, checked the applciation and pressed the mechanical Play button ▶ on the Smartpad.

Within 10 seconds before the timeout, we launch: **roslaunch iiwa_driver iiwa_bringup.launch.** 

**Problem:** There seems to be a problem with the connection between the FRIOverlay as the Smartpad does not provide us with a selection for the control mode and stiffness. It merely shows a timeout (see SmartPAD.jpg:
[image]

And the launch file seems to stall halfway without publishing > /iiwa/joint_states:
[image]

**Next step:** Any thoughts on where this error might originate from would be appreciated. 

NB. We are noobs with ROS.

costashatz, 2022-03-28:
@cmheunis did you solve this?

If not, this is an error coming most likely for not configuring your robot correctly. Have you configured the robot correctly (see [here](https://github.com/epfl-lasa/iiwa_ros#sunrise-robot-application-upload))? Also the configuration of the linux laptop (see [here](https://github.com/epfl-lasa/iiwa_ros#bringup-iiwa_driver)) is important. The ROS_MASTER_URI in your picture seems wrong. See also this [issue](https://github.com/epfl-lasa/iiwa_ros/issues/14).

costashatz, 2022-03-28:
There are two different computers that need setup: one that has the Sunrise (on Windows) and one that will run ROS (on Linux).

CodingCatMountain, 2023-04-20:
> There are two different computers that need setup: one that has the Sunrise (on Windows) and one that will run ROS (on Linux).

Hi, @costashatz . It must be two compuaters? Could dual systems(Windows and Ubuntu on the same compuaters) work?

matthias-mayr, 2023-04-20:
In principal you can also dual-boot, but then one has to get the network setup correct, so IPs and the correct cable connections.

Most likely the machines can not reach each other with the network connection for FRI. You can for example try to ping your robot on that IP address. You can also connect a screen to your cabinet and try to ping your host computer.

Sometimes we also have connectivity issues in a setup that is configured correctly and in that case it helped to reboot the machine.

closed 2023-04-28 by costashatz