epfl-lasa/iiwa_ros #83: fatal error: pluginlib/class_list_macros.hpp: No such file or directory

victoralad, 2022-04-26:
It appears that the customEffortController was not added as an executable in the CMakelist.txt file. When I added it and tried to compile using `catkin build iiwa_ros`, I get the error `fatal error: pluginlib/class_list_macros.hpp: No such file or directory`. Is this something that anyone else has run into?

victoralad, 2022-05-02:
I figured out that the customEffortController does not need to be added as an executable. To use the controller, I just needed to add it to my iiwa_control.lauch file

closed 2022-05-02 by victoralad

costashatz, 2022-05-03:
Ah yes! You do not need to add it as an executable. It is a plugin to be loaded. Sorry for not answering your issue. I was on vacations.