epfl-lasa/iiwa_ros #84: Robot does not hold current position when iiwa_driver is launched

victoralad, 2022-05-02:
According to the instructions on this repository, to launch the robot, the application is selected on the smartpad and the command `roslaunch iiwa_driver iiwa_bringup.launch` is run in the terminal. However, when I do that, and select `Torque` (the other options are `wrench` and `position`) on the smart pad, the robot suddenly starts moving very fast to a configuration I have no idea what it is. I expected the robot to stay in its current pose. Am I missing something or is there a solution to this?

costashatz, 2022-05-03:
Which controller are you using? The [default one](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_driver/launch/iiwa_bringup.launch#L33) or the `CustomEffortController`?

victoralad, 2022-05-03:
Yes, I used the default one. Are you suggesting I use `TorqueController` instead? Based on how the default controller behaves in simulation, I suspect there is no gravity compensation being done by the `TorqueController`. However, the motion I observed on the hardware appears to be an active torque commanded, and not just the lack of gravity compensation.

victoralad, 2022-05-03:
I went ahead to try the `CustomControllers` on the hardware, but I got the same behavior as I did when I used `TorqueController`. The robot suddenly starts moving to an unknown configuration.

costashatz, 2022-05-04:
This is weird. The TorqueController should do nothing (just gravity compensation). Are you sure that you are following the bringup instructions [here](https://github.com/epfl-lasa/iiwa_ros#bringup-iiwa_driver) correctly?

The only other thing that comes to my mind is the following: Do you have some tool attached to the end-effector? Which Java file have you loaded (look [here](https://github.com/epfl-lasa/iiwa_ros/tree/master/iiwa_driver/java))? `FRIOverlay` or `FRIOverlayGripper`? If you have weight attached to the end-effector without launching the gripper file or you have not configured your gripper correctly (you need to identify the mass properties) then you might get weird behaviors.

victoralad, 2022-05-04:
Hmm, I may have used `FRIOverlay`. Indeed, I have a Robotiq gripper at my end effector. I may need to switch to FRIOverlayGripper. I will try it out and let you know how it goes.
Thanks

matthias-mayr, 2022-05-05:
As you probably found out, the line in `FRIOverlayGripper`
```java
_toolAttached = getApplicationData().createFromTemplate("MyVirtualGripper");
```
needs to have the name of the tool that is known to the KUKA Java stack. But then - besides some slight miscalibration it works for us.

victoralad, 2022-05-06:
Indeed, using `FRIOverlayGripper` when I have a gripper attached works after proper calibration. When I use `TorqueController`, the robot holds its position. However, when I use `CustomControllers`, the robot still tries to go to an unknown location. I went back into the code [custom_effort_controller.cpp](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/src/custom_effort_controller.cpp) to review what might be going on. However, I could not tell for sure if the control is a full task space control (position + orientation) or if it is just `orientation` control. I know the default is a `task-space` control, but the `iiwa_control.yaml` file currently sets the type to just orientation (see [CustomControllers type](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_control/config/iiwa_control.yaml#L80)). 

In summary, what exactly is the `CustomControllers` doing?

Lastly, it appears there is no consideration of Coriolis forces. Was this done on purpose?

Thanks

victoralad, 2022-05-08:
@costashatz I tested the CustomControllers again, and it works! The problem I had before was that I had modified the custom_effort_controller.cpp file to account for my namespace modification. This resulted in the weird behavior I saw. Once I fixed that, everything appears to be working perfectly for just a single robot. 

I am now working on addressing the issues with namespace as recommended by you and Matthias in the other post ([namespace](https://github.com/epfl-lasa/iiwa_ros/issues/85)). It appears wrapping namespace for simulation seems to be slightly different from wrapping namespace for the hardware. On the hardware, it is as simple as creating two yaml files for the controller and passing the appropriate yaml file for each robot. Whereas in simulation, I also have to worry about spawning both robots, so I must pay attention to how I setup the `robot_description` parameter. I was hoping I could load just one `robot_description` to the parameter server, and spawn two separate robots using the same `robot_description`, but it appears that is not possible.

costashatz, 2022-05-09:
> @costashatz I tested the CustomControllers again, and it works! The problem I had before was that I had modified the custom_effort_controller.cpp file to account for my namespace modification. This [...]

Happy that this is working..

> I am now working on addressing the issues with namespace as recommended by you and Matthias in the other post ([namespace](https://github.com/epfl-lasa/iiwa_ros/issues/85)). It appears that wrapping namespace for simulation seems to be slightly different from wrapping namespace for the hardware. On the hardware, it is as simple as creating two yaml files for the controller and passing the appropriate yaml file for each robot. Whereas in simulation, I also have to worry about spawning both robots, so I must pay attention to how I setup the `robot_description` parameter. I was hoping I could load just one `robot_description` to the parameter server, and spawn two separate robots using the same `robot_description`, but it appears that is not possible.

Let's move this discussion in #85 . See @matthias-mayr's reply. Closing this.

closed 2022-05-09 by costashatz

victoralad, 2022-06-15:
Hello Konstantinos,
I would like to cite iiwa_ros, please do you have a bibtex reference for your software?
Regards

Victor Aladele

PhD student

Robotics and Machine Learning

Electrical and Computer Engineering Department

Georgia Institute of Technology

The true sign of intelligence is not knowledge but imagination.

Albert Einstein<http://www.brainyquote.com/quotes/authors/a/albert_einstein.html>

costashatz, 2022-06-28:
> Hello Konstantinos, I would like to cite iiwa_ros, please do you have a bibtex reference for your software? Regards

Check [here](https://github.com/epfl-lasa/iiwa_ros#citing-iiwa_ros). This should suffice for now. Aiming for a proper citation soon :)

victoralad, 2022-06-30:
Great! Thank you!
Regards

Victor Aladele

PhD student

Robotics and Machine Learning

Electrical and Computer Engineering Department

Georgia Institute of Technology

The true sign of intelligence is not knowledge but imagination.

Albert Einstein<http://www.brainyquote.com/quotes/authors/a/albert_einstein.html>

XinyuanZhao, 2023-02-01:
> According to the instructions on this repository, to launch the robot, the application is selected on the smartpad and the command `roslaunch iiwa_driver iiwa_bringup.launch` is run in the terminal. [...]

Hello. I met the same problem as described here when I managed to communicate with the robot and run `roslaunch iiwa_driver iiwa_bringup.launch`. The robot didn't hold its current position but tried to move to another configuration.The following are what I found and checked.
- There's no gripper attached to the flange (see the picture below).
- When I set the stiffness to be 50 or 100, the robot moved fast. When I set the stiffness to be 0 (so there should be only the automatic gravity compensation), the robot moved gently but it still tried to reach another configuration.
- At the configuration it tried to reach the robot had its joint_4 and joint_6 quite bent to the limits (see the picture below).

Do you have any ideas why it is the case? 
By the way, how does the controller decide the reference joint positions when I launch the `TorqueController`?
[image]

costashatz, 2023-02-01:
> * When I set the stiffness to be 50 or 100, the robot moved fast. When I set the stiffness to be 0 (so there should be only the automatic gravity compensation), the robot moved gently but it still tried to reach another configuration.

This seems to me like a bad calibration of the robot. Try to calibrate the robot using your smartpad.

> how does the controller decide the reference joint positions when I launch the `TorqueController`?

The `TorqueController` does not have reference joint positions. It waits for torque commands looking at the topic `/iiwa/TorqueController/command` (assuming you have all defaults namespaces etc).

> Do you have any ideas why it is the case?

This is most probably due to wrong configuration. Are you using `FRIOverlay` or `FRIOveralyGripper`?

costashatz, 2023-02-01:
> It is worth mentioning that the robot keeps still when launching the torque controller in all the cases.

Wait. What do you mean here? I thought you were saying that it wasn't working with the torque controller.

> So basically we suppose that the default URDF used by KUKA to realize the gravity compensation is not perfect -- the modeling mismatch may cause larger effect on the compensation under some joint configuration than under others.

Yes, the gravity compensation is not perfect. It doesn't account for velocities for example. It's pure gravity compensation. Also KUKA is not using a URDF and their model is pretty good. It's just that you cannot compensate for everything.

> But anyway we are able to achieve simple joint-wise torque control using IIWA_ROS.

Your signal should be much stronger than the gravity compensation in any case.

XinyuanZhao, 2023-02-01:
> Wait. What do you mean here? I thought you were saying that it wasn't working with the torque controller.

Sorry for the misleading expressions. I wanted to say that the **joint velocities are zero** when launching the torque controller so that the gravity compensation was not affected by initial movement of the robot.

> It doesn't account for velocities for example. It's pure gravity compensation.

Thank you for the reminder. Previously I suppose KUKA also took Coriolis and centrifugal into account.

> It's just that you cannot compensate for everything.

It is the case. I cannot adjust the performance of gravity compensation by modifying the URDF accordingly.

matthias-mayr, 2023-02-01:
We typically run a [Cartesian impedance controller](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) with iiwa_ros. Then it is not that apparent, except if we completely disable the stiffness.

XinyuanZhao, 2023-02-02:
> We typically run a [Cartesian impedance controller](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) with iiwa_ros. Then it is not that apparent, except if we completely disable the stiffness.

Thank you again for sharing this toolbox, which seems to be a well-designed impedance controllers for manipulator's control. We will follow it later if we find it useful for our study.

Just think of another question using `iiwa_ros`: what are the stiffness gains (e.g. 0, 50, 100, etc.) shown on SmartPad when I just run `roslaunch iiwa_driver iiwa_bringup.launch`? Are they joint-wise stiffness? Am I able to tune these values in my controller via `rostopic` or `rosservice`?

matthias-mayr, 2023-02-02:
> Are they joint-wise stiffness?

Yes, you are configuring KUKA's joint impedance controller

> Am I able to tune these values in my controller via rostopic or rosservice?

Not really. Once the FRI connection is built up, only joint position references and joint effort references are sent from the ROS side.

There is the [code around grl](https://github.com/ahundt/grl) that allowed to configure things on the robot side with a different channel. I spent some time with it in 2020, but did not make it work before I opted for another solution. It seems like the repository got abandoned when the PhD student left.

XinyuanZhao, 2023-02-02:
> Yes, you are configuring KUKA's joint impedance controller

Thanks! But what is the reference joint position for KUKA's joint impedance controller? Am I able to check and modify this value? Are there also default damping gains for that joint impedance controller?

And further, if I implement another controller and publish torque signal, I wonder if the robot will receive a compound torque command of _default gravity compensation_ + _default joint impedance control_ + _user's control_?

matthias-mayr, 2023-02-02:
> Thanks! But what is the reference joint position for KUKA's joint impedance controller?  Am I able to check and modify this value? 
When it's initialized it's the current position & later on whatever ROS sends. You can switch to the `PositionJointInterface` in `iiwa_ros` for that & then use whatever controller.

> Are there also default damping gains for that joint impedance controller?

I assume so. That is on the KUKA side. You should have received some documentation for their Java API

>  I wonder if the robot will receive a compound torque command of default gravity compensation + default joint impedance control + user's control?

FRI allows users to overlay KUKA's control with custom references. And this is what is sent from the ROS side with `iiwa_ros`. KUKA's control is gravity compensation + the control signal of the controller that is started on the Java side. (which is the Joint impedance for you right now)

This means that if some stiffness for that  joint impedance controller is defined, then it would output a torque signal as well. When we do torque control like in the Cartesian impedance controller repo we typically do not want that the KUKA controller to interfere, so we set the stiffness to 0.

The gravity compensation would always be applied and the it's _gravity compensation + user torques_

XinyuanZhao, 2023-02-02:
> You can switch to the `PositionJointInterface` in `iiwa_ros` for that & then use whatever controller.

Glad to know that I may have control over the reference joint position for KUKA's joint impedance controller. May you please explain this operation in a bit more detail? Some questions are listed below for example
- If I set `PositionJointInterface` in `iiwa_setup.launch`, do I have to change the controller accordingly (to `PositionController`)?
- If I launch the robot with the `PositionJointInterface` option, am I able to publish torque command to `/iiwa/TorqueController/command` later? (as you said "use whatever controller")

> When we do torque control like in the Cartesian impedance controller repo we typically do not want that the KUKA controller to interfere, so we set the stiffness to 0.

It is very likely that we will also set the stiffness to 0 so as to avoid interference between KUKA's controller and our custom controller. I want to play with KUKA's stiffness gains and its reference joint position mainly because the robot didn't behave as we expected when I gave non-zero stiffness: the robot moved rapidly to an unknown configuration and incur safety shutdown (while with 0 stiffness the robot seems to be a bit normal). So there should be something wrong such as incorrect configuration from us or mechanical failure of one joint, etc. Hope I can figure them out before they ruin the results in the future.

anubhav-dogra, 2023-04-03:
To do the proper load data calibration, Is there any optimal configuration for the robot to start with the Load Data Determination. KUKA manuals only talk about A5, A6 and A7, but I have found significant variations in the result in different configurations (A1, A2, A3 and A4). 
Can (@matthias-mayr @costashatz ) you please comment on this?