epfl-lasa/iiwa_ros #98: Segfault upon URDF parsing with iiwa_gazbo

niederha, 2023-01-18:
Hello,

I have installed iiwa_ros and all the dependecies in a docker with ros melodic and gazebo9. However as soon as I launch `roslaunch iiwa_gazebo iiwa_gazebo.launch` I get the following output:

```
...
process[iiwa/iiwa_service-2]: started with pid [2655]
[ INFO] [1674030234.814907200]: Starting Iiwa IK server..
[ INFO] [1674030234.817402241]: IiwaService is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1674030234.917996468]: Received urdf from param server, parsing...
[iiwa/iiwa_service-2] process has died [pid 2655, exit code -11, cmd /home/ros/ros_ws/devel/lib/iiwa_tools/iiwa_service __name:=iiwa_service __log:=/home/ros/.ros/log/702a2b8c-9709-11ed-a5d3-0242ac110002/iiwa-iiwa_service-2.log].
log file: /home/ros/.ros/log/702a2b8c-9709-11ed-a5d3-0242ac110002/iiwa-iiwa_service-2*.log
...
```

I have tried to track down the reason and it seems that the URDF parsing done in `IiwaTools::init_rbdyn` with `_rbdyn_urdf = mc_rbdyn_urdf::rbdyn_from_urdf(urdf_string);` (iiwa_tools.cpp, line 369) yields a segfault.

As far as I could check the URDF seems valid, I've been able to open it and parse it with a different python tool. 

I'll keep researching and posting here as I investigate. Any pointers are appreciated.

costashatz, 2023-01-18:
This "smells" like a native flag issue. Make sure that you compile all dependencies with the same native flags. Either enable or disable all SIMD flags. The current master branches of iiwa_ros and https://github.com/epfl-lasa/robot_controllers compile everything without SIMD flags. See [here](https://github.com/epfl-lasa/iiwa_ros#kuka-fri-private) for compiling kuka-fri without the SIMD flags. I also just updated the README to compile RBDyn and SpaceVecAlg without native flags: so have a look [here](https://github.com/epfl-lasa/iiwa_ros#spacevecalg), [here](https://github.com/epfl-lasa/iiwa_ros#rbdyn) and [here](https://github.com/epfl-lasa/iiwa_ros#mc_rbdyn_urdf) to compile/install those correctly.

niederha, 2023-01-18:
Good nose, it was indeed a SIMD flag issue. 
I think I didn't get that the patch on kuka_fri was removing the SIMD flags and not adding them. 
But indeed by making sure that the flags are either activated or deactivated everywhere everything runs smoothly.
Thanks for the pointer!

closed 2023-01-18 by niederha