epfl-lasa/iiwa_ros #100: End effector buttons and light?

kowac, 2023-03-03:
Checked the repo but haven't found any answer within the code:
1. Is it possible to change the end effector's LED ring light from within ROS?
2. Is it possible to get the end effector's buttons (originally for handguiding) to trigger some signal within ROS?

If these two functions can only be used in KUKA's own ecosystem, this question can be closed. I just thought that asking could not hurt.
Thanks.

matthias-mayr, 2023-03-04:
I am sure that this is not implemented in this driver. afaik this is also nothing that can be done with `FRI`, but I didn't pay attention to this when going through the FRI documentation.

If you really need this, then [iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack) might be an easier starting point. However it seems to be unmaintained for a couple of years now.

kowac, 2023-03-04:
That's what I thought. Thank you for the quick reply!

closed 2023-03-04 by kowac