epfl-lasa/iiwa_ros #104: More details OR a case about how to utilize the TorqueController or PositionController?

CodingCatMountain, 2023-04-27:
Hi, my respected developers. After I have set up the iiwa_ros on the real iiwa robot in our lab, I get confused to how to properly utilize this package. I wonder that could you share some basic case about how you use it? Are there any videos or papers which utilized this package? I will appreciate it for any advise or instructions.

CodingCatMountain, 2023-04-28:
Thank you for the sharing! @costashatz .

closed 2023-04-28 by CodingCatMountain

matthias-mayr, 2023-05-14:
What you are configuring is the joint impedance control mode. A stiffness of 20 is quite low. If you set it to 0, the robot would not move at all (except for calibration errors). There is also a real position control mode that is not in the Java example file.