epfl-lasa/iiwa_ros #108: Inverse Dynamic control and gravity compensation problem

Shravista, 2023-06-04:
Hii,

We are trying out the inverse dynamics controller for the kuka using the mass matrix and velocity product obtained from the `/iiwa/iiwa_mass_server` and `/iiwa/iiwa_gravity_server` respectively. In the case of the velocity product, the request to the server is with gravity vector zero and the torques field is the current effort coming from `/iiwa/joint_states`.  The controller looks like this below

[image]

The gravity server requires torques as inputs. why do we need to give torque when requesting the `/iiwa/iiwa_gravity_server` for the response? Further, what torques do we need to provide, the commanded torques or the torques including gravity compensation?

costashatz, 2023-07-14:
First, apologies for the late answer, but I was traveling a lot the last month and didn't get the chance to reply.

> The gravity server requires torques as inputs. why do we need to give torque when requesting the /iiwa/iiwa_gravity_server for the response?

This is wrong. Most probably because of some copy-paste. Let me fix that.

> is the below ros graph correct?

The graph and the controller looks ok.

Did you manage to make it work?

Shravista, 2023-07-14:
Currently, I couldn't make it work and on top of that kuka robot in our lab is under maintenance. So once the maintenance is completed I will test it out. Additionally, I think that mass and inertia matrix values are not so close to the actual robot, I know a way to figure out the same or use the Adaptive Torque controller (similar to the inverse dynamics controller) from Bruno Scillano's book and see the tracking. I will let you know once our robot's maintenance is complete and we tested our controllers.