epfl-lasa/iiwa_ros #109: Catkin_make failed.

qwertyuiop789456, 2023-07-14:
**I have added all the packages as per instruction. But when i do catkin_make i get the following error in the terminal at 98 percent.**
I have attached an google doc as the error goes up to 33 pages. I have attached a snapshot of first and end of the error.

https://docs.google.com/document/d/1M8b-NZ5g3CUhBh0zA0FNuWTBxQiMYdSvkMQtOSvqqbo/edit?usp=sharing

[image]

[image]

costashatz, 2023-07-15:
@qwertyuiop789456 thanks for using the library.

Which ROS version are you using? And which Gazebo version? How did you install Gazebo?

From the message, the error should be fixed if you change [this line](https://github.com/epfl-lasa/iiwa_ros/blob/master/iiwa_gazebo/CMakeLists.txt#L51) to (C++14 or above):

```CMake
set_property(TARGET iiwa_gazebo_gravity_compensation_hw_sim PROPERTY CXX_STANDARD 14)
```

But I am speculating :)

qwertyuiop789456, 2023-07-17:
Hi @costashatz . sorry for not mentioning the details. 

ROS - Noetic
Gazebo - 11.11.0

Gazebo was installed when i used "sudo apt install ros-noetic-desktop-full"

Even after changing to 14, still the error remains.

[image]

costashatz, 2023-07-17:
> ROS - Noetic
> Gazebo - 11.11.0

This is the issue. Gazebo and ROS in general changed to using the `std` lib instead of `boost` at `Noetic`. At the moment, the recommended version (as said in the docs) is to use `Melodic` or `Kinetic`. The package has been extensively used and tested in those versions.

Let me put a TO-DO to make the package compatible with `Noetic` as well. Please switch to `Melodic` until I make it compatible with `Noetic`.

Closing this issue.

closed 2023-07-17 by costashatz