epfl-lasa/iiwa_ros #111: What FRI versions are compatible with the ROS driver?

ian-chuang, 2023-11-02:
I'm looking to buy a used KUKA iiwa 14 R820 and control it with this ROS driver.

I see that there are multiple versions of FRI (1.16, 1.17, and 2.5) so I was wondering if the version of FRI matters for the ROS driver. Has FRI v2.5 been tested?

Also, I'm just curious if I am forced to pay the hefty price of $6,000 just to get the FRI software and control the robot with ROS. Is there any other way to get FRI?

matthias-mayr, 2023-11-02:
> Has FRI v2.5 been tested? Does all the KUKA iiwa arms come with FRI

We have 1.15 and did not see a reason to upgrade. We know that we could not install older versions.

> Does all the KUKA iiwa arms come with FRI?

My understanding would be that all normal iiwas come with the possibility to do FRI. Not sure about the iiwa med.

> Is there any other way to get FRI? Can I just contact KUKA and they will provide me with FRI or Sunrise Connectivity Suite?

No idea. We just had it as an options package that we install with Sunrise Workbench. You can of course try to reach out to them. I didn't have a lot of contact with them.

ian-chuang, 2023-11-03:
Thanks a lot for the quick answer. 

I contacted KUKA for a quote on a used iiwa 14 and FRI is a paid add-on ($3000 if bought with their used robot). Though the quote recommended buying FRI v1.7 :man_shrugging:

I want to buy a KUKA iiwa and use it primarily with ROS for research purposes, but I'm a bit hesitant with these potential compatibility issues. It also seems like a bit of a pain to setup with ROS.

closed 2023-11-03 by ian-chuang